zwave-js/zwave-js #6144: Controller 700 problem after update - all nodes are gone

luzik, 2023-08-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

- Upgrade

What do you observe?

- UI do not show any node, nor home id

### Device information

Manufacturer: Aeonlab
Model name:  stick 700

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.22.3.8607bd8
zwave-js: 11.9.2
home id: undefined
home hex: undefined

### Did you change anything?

no

### If yes, what did you change?

docker upgrade

### Did this work before?

Yes (please describe)

### Attach Driver Logfile

[zwavejs_2023-08-10.log](https://github.com/zwave-js/node-zwave-js/files/12312324/zwavejs_2023-08-10.log)

AlCalzone, 2023-08-10:
You seem to have soft-reset disabled in the Z-Wave JS UI config. Please enable it, then try again.

luzik, 2023-08-10:
Yeah it helped, thx.
I found this in troubleshooting, but I am using docker with usbip protocol over network so was quite scare to turn this on.
Maybe it is good idea to leave this issue open for a week or two

AlCalzone, 2023-08-10:
700/800 series controllers have no issues with soft-reset. The reason we never turned it on by default were some 500 series sticks which can reconnect afterwards, or need to be manually re-plugged in the worst case.

closed 2023-08-14 by zwave-js-bot