zwave-js/zwave-js #6147: Missing device configuration: SHELLY WAVE 1PM

chma73, 2023-08-11:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

SHELLY WAVE 1PM

### Manufacturer ID

0x0460

### Product Type

0x0002

### Product ID

0x0084

### Firmware Version

10.13.0

### Do you have a manual?

[Wave_1PM_multilang_2023_v06.pdf](https://github.com/zwave-js/node-zwave-js/files/12322496/Wave_1PM_multilang_2023_v06.pdf)

33Fraise33, 2023-09-06:
I found the following files on other places:
https://opensmarthouse.org/zwavedatabase/1577 => #6147
https://opensmarthouse.org/zwavedatabase/1578 => #6214
https://products.z-wavealliance.org/products/4922?selectedFrequencyId=-1 => No issue created yet

These are the 3 new devices that Shelly has brought out. I am unsure how to get started on this but I just wanted to let you know.

AlCalzone, 2023-09-13:
@zwave-js-bot import config 4922 from zwa

33Fraise33, 2023-09-16:
Today I onboarded the shelly devices. All the values seem to be there except for Manufactur and Product. 
[image]
[image]

33Fraise33, 2023-09-16:
Shelly also already has documentation on HA and Zwave JS: https://support.shelly.cloud/en/support/solutions/folders/103000553456

AlCalzone, 2023-09-21:
These appear to be the auto-discovered config parameters. Someone from Shelly/Qubino is working on a cleaned up config file here: https://github.com/zwave-js/node-zwave-js/pull/6289

PR #6280 (2023-09-21): feat(config): add Shelly Wave 1PM
fixes: #6147

**TODO:**
- [x]	Change PR title to be more specific
closed 2023-09-21 by AlCalzone