zwave-js/zwave-js #6150: Zooz ZST39 Error: DRIVER   Controller identification returned invalid node ID 0

cvocvo, 2023-08-13:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I have a new Zooz ZST39 800 series usb stick and I'm trying to configure Home Assistant to work with it. Last night I set it up and paired a Zcombo smoke alarm and I thought everything was great. However, overnight my Home Assistant did a weekly reboot and this morning I checked and Zwave wasn't working -- it couldn't connect to the Zwave JS Addon. I thought that was odd; I tried:
- reloading home assistant
- rebooting
- cold booting
- removing and trying to readd zwave

Nothing seemed to work. Then I tried moving USB ports and I was able to setup Zwave again. However after a reboot I was met with the same error.
>DRIVER   Controller identification returned invalid node ID 0
The zwave js addon just restart-loops trying to fix itself but has the same error.

What do you observe?
(see above)

What did you expect to happen?
Zwave to work after reboots

Steps to reproduce the behavior:
Not sure how; see above description.

### Device information

Manufacturer: Zooz
Model name: ZST39
Node ID in your network: Unknown, Zwave thinks it's 0 but it was 1 before I rebooted and everything broke.

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 11.9.2 DRIVER, 11.9.1 CONFIG
Home assistant addon version: Current version: 0.1.86
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Worked the first time and stopped working after a reboot.

### Attach Driver Logfile

[log file.txt](https://github.com/zwave-js/node-zwave-js/files/12329645/log.file.txt)

(I don't know how to get the log file off of home assistant -- I turned it up to debug level logging and copy-pasted the logs out to a text file.)

cvocvo, 2023-08-13:
> 👋 Hey @cvocvo!
> 
> It looks like you attached a logfile, but its filename doesn't look like it a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) that came from Z-Wave JS. Please make sure you upload the correct one.

Sure -- how do I get that from the home assistant Z-Wave JS addon?

jraynes, 2023-08-13:
I just had exactly the same problem. Noticed my HA automation to turn off some lights at sunrise did not work this morning. All zwave entities in HA were greyed out and the Zwave JS addon was crashing over and over with error "DRIVER Controller identification returned invalid node ID 0". Tried enabling debug mode to grab logs but I could not find an option to export to file.

I had to stop the the add on, physically unplug the zwave stick, plug it back in, start the add on and zwave is working again.

Similar to @cvocvo I have an automation that reboots HA every Sunday morning and that appears to be when the problem started. Maybe Zwave JS didn't shut down cleanly?

cvocvo, 2023-08-13:
> I just had exactly the same problem. Noticed my HA automation to turn off some lights at sunrise did not work this morning. All zwave entities in HA were greyed out and the Zwave JS addon was [...]

Confirmed that unplugging and replugging the USB allows it to start up fine. Good call!

cvocvo, 2023-08-13:
> Downgrade your add-on or switch to Z-Wave JS UI if you want a solution right now. [home-assistant/core#98250](https://github.com/home-assistant/core/issues/98250)
> 
> Same issue: #6148

Oddly I downgraded via instructions here: https://github.com/home-assistant/core/issues/98250#issuecomment-1674616357
Then rebooted HA and I’m back to the same error again. 
Unplugging and replugging the USB seems to fix it again.

PR #6153 (2023-08-14): fix: change order of startup commands to consider 16-bit node IDs
This PR changes the order of some commands during startup/controller identification in order to parse commands correctly if the controller is already in 16-bit node ID mode.

fixes: #6150
fixes: https://github.com/zwave-js/node-zwave-js/issues/6144
fixes: https://github.com/zwave-js/node-zwave-js/issues/6148

for: https://github.com/home-assistant/core/issues/98250
for:
closed 2023-08-14 by zwave-js-bot