zwave-js/zwave-js #6160: Swidget subparameters are not being handled correctly

michaelkkehoe, 2023-08-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
When trying to set sub-parameters on Swidget devices via either the ZWaveJS-UI or through Home Assistant, the write action fails

What do you observe?
Setting of sub-parameters fails, but regular parameters succeed. It appears that the value-size may be incorrect (see logs).

We (myself and the manufacturer believe the following is happening):
1. We can see it is sending with a value size of 1 when setting a sub-parameter 14, which is incorrect.
2. We can also see in the log when it does a Get request: The response is correctly parsed as a value size of 2:

What did you expect to happen?
The setting of the sub-parameter should succeed

Steps to reproduce the behavior:
N/A

### Device information

Manufacturer: Swidget
Model name: ZW006UWA
Node ID in your network: 3

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:11.9.2
`zwave-js-ui` branch: 8.22.3

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have manually set the sub-parameter with the zwave-js-ui by setting a bit-mask, but have been unable to set the sub-parameter using any other UI option

### Attach Driver Logfile

```
2023-07-07T20:24:43.569Z DRIVER » [Node 003] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      4
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 8
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x2f6072f989e5f006d1e98815295f160f
                                    │ security class:  S2_Authenticated
                                    └─[SupervisionCCGet]
                                      │ session id:      59
                                      │ request updates: true
                                      └─[ConfigurationCCSet]
                                          parameter #:      14
                                          reset to default: false
                                          value size:       1
                                          value format:     UnsignedInteger
                                          value:            98
2023-07-07T20:24:44.639Z DRIVER » [Node 003] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      5
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 9
                                    │ security class:  S2_Authenticated
                                    └─[ConfigurationCCGet]
                                        parameter #: 14
2023-07-07T20:24:44.690Z DRIVER « [Node 003] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -66 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 45
                                    │ security class:  S2_Authenticated
                                    └─[ConfigurationCCReport]
                                        parameter #: 14
                                        value size:  2
                                        value:       99
```

michaelkkehoe, 2023-08-14:
The device configuration guide may also be helpful: https://files.swidget.com/Integrations/UserManuals/Swidget_Z-Wave_Insert_Configuration.pdf

AlCalzone, 2023-08-14:
FWIW, I simulated a fake ZW008UWA that just has the parameter 14, no security, no supervision. When I change a partial, this command is sent:
```
21:10:39.661 SERIAL » 0x010d0013020670040e020062255d87                                    (15 bytes)
21:10:39.662 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      93
                      └─[ConfigurationCCSet]
                          parameter #:      14
                          reset to default: false
                          value size:       2
                          value format:     UnsignedInteger
                          value:            98
```

michaelkkehoe, 2023-08-14:
On a different node (ZW001UWA) with S2 security:
```
2023-08-14 12:49:12.764 INFO Z-WAVE: Calling api writeValue with args: [
  {
    nodeId: 114,
    commandClass: 112,
    endpoint: 0,
    property: 14,
    propertyKey: 8
  },
  1,
  null,

  [length]: 3
]
2023-08-14 12:49:12.772 INFO Z-WAVE: Writing 1 to 114-112-0-14-8
2023-08-14 12:49:12.983 ERROR Z-WAVE: Unable to write 1 on 112-0-14-8: Fail
2023-08-14 12:49:12.987 INFO Z-WAVE: Success zwave api call writeValue { status: 2 }
```

michaelkkehoe, 2023-08-14:
My apologies. After forcing a re-interview, things seem to work as expected. I'll ask the contact at the manufacturer to test and confirm before closing this

michaelkkehoe, 2023-08-19:
Confirmed that re-interviewing the device fixes the issue. Apologies for the false issue.

closed 2023-08-19 by michaelkkehoe