zwave-js/zwave-js #6164: Ring 4AK1SZ

candrea77, 2023-08-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? I can't get any event in zwave / Home assistant
I've noticed that I have no assciation .... any GROUPS is missing

What do you observe? In the zwavejs ui , under GROUPS there is "no data avilable".
If I try to add an association, nothing happen, still "No data available" message.

What did you expect to happen?  Add an association with NodeID_1

Device is configured with "S2 Access Control"

I've tryed the "Re-interview Node" function but still no success

### Device information

Manufacturer: Ring
Model name: 4AK1SZ
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.22.3.8607bd8
zwave-js: 11.9.2

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

2023-08-14 23:15:55.722 DEBUG SOCKET: Event ZWAVE_API emitted to j7fvQDdDBm9594VfAAAq
2023-08-14 23:15:55.723 INFO Z-WAVE: Calling api addAssociations with args: [
{ nodeId: 157, endpoint: 0 },
1,
[ { nodeId: 1, endpoint: 0 }, [length]: 1 ],
[length]: 3
]
2023-08-14 23:15:55.724 INFO Z-WAVE: [Node 157] Adding Node 1 to Group 1 of Node 157
2023-08-14 23:15:59.284 INFO Z-WAVE: Success zwave api call addAssociations true
2023-08-14 23:15:59.313 DEBUG SOCKET: Event ZWAVE_API emitted to j7fvQDdDBm9594VfAAAq

AlCalzone, 2023-08-15:
What the bot said

candrea77, 2023-08-15:
Yep,.I'll send the log as soon as i came back to home.

Do you need log when I'll try to made the group association ?
Or something different ?

Mail inviata da dispositivo Mobile

AlCalzone, 2023-08-15:
Ideally include a re-interview and the failed attempt at creating association in the log.

candrea77, 2023-08-16:
Hi there,
I've also take time to git pull the latest release (zwavejs/zwavejs2mqtt:master)

Now I'm on :
zwave-js-ui: 8.22.3.62ecc4b
zwave-js: 11.11.0

Below you can find the log (start zwavejs + Re-interview Node)

Let me know in case I've to do something else.

[zwavejs_2023-08-16.zip](https://github.com/zwave-js/node-zwave-js/files/12362468/zwavejs_2023-08-16.zip)

candrea77, 2023-08-16:
And sorry .... I forgot the reassociation.
Here you are the related log
[zwavejs_2023-08-16.zip](https://github.com/zwave-js/node-zwave-js/files/12363808/zwavejs_2023-08-16.zip)

AlCalzone, 2023-08-17:
It looks like you are having connectivity issues. Lots of timeouts during the interview. Sending commands takes a long time and there's a ton of permanent traffic from sensors.

I have a few links I suggest working through, then try again:
1. https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
1. https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=optimizing-the-reporting-configuration
1. https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength

---

FWIW, associations work fine if there is no connectivity problem:
[image]

candrea77, 2023-08-17:
Hi there,
I moved the Ring KeyPad very close to the controller and restart interview just to exclude any possibile zwave network problem (same room , 1 meter distance)

After reinterview finished, no group association is present.

I've tried again manual group association, but nothing changed.
Here it is the new logs ....

[zwavejs_2023-08-17.log](https://github.com/zwave-js/node-zwave-js/files/12369905/zwavejs_2023-08-17.log)

BTW : Another strange thing ..... I've got no battery indicator ('null%' , in your screenshoot , it seems ok on your device)
[image]

Wich version of Firwmware is yours (mine 1.18.0 sdk 7.12.4) ?

May I've to try to delete the device and reiinclude it in order to solve my problems ?

AlCalzone, 2023-08-17:
```
2023-08-17T13:22:16.274Z CNTRLR » [Node 157] querying battery status...
...
2023-08-17T13:22:17.344Z CNTRLR   [Node 157] Timed out while waiting for a response from the node (ZW0201)
```
For some reason your keypad is not responding to that query. It doesn't seem to be a communication issue, as it does confirm that it gets the commands.

Mine has the same firmware as yours. Have you tried putting it on the USB cable for the interview yet? Maybe the load of queries is a bit too much for it on battery power.

candrea77, 2023-08-17:
> Have you tried putting it on the USB cable for the interview yet? Maybe the load of queries is a bit too much for it on battery power.

Thanks for quick reply.
Unfortunately It's already on USB power.

AlCalzone, 2023-08-17:
Do you have a different Z-Wave stick you could use for a test? Otherwise I'm out of ideas. I don't have any issues, neither on USB or battery power.

candrea77, 2023-08-17:
Unfortunately no.

Just a quick question .... if I get a different stick can i backup and restore the whole zwave db (all devices) from old to new one ?
I'm planning to switch to a 700chipset, and that's can be a valid reason to speed-up the change.

Meanwhile, I'll try exclude and re-include and let you know.

AlCalzone, 2023-08-17:
With your stick and firmware version, yes. Just do an NVM backup in Z-Wave JS UI, and restore that on the new stick.

If you're located in EU, you should know though that the Z-Stick 7 has severe RF range issues for an unknown reason. I've had the best performance with 500 series controllers and those 700 series controllers that you put on the GPIO of a Raspberry Pi (or similar).

candrea77, 2023-08-17:
> With your stick and firmware version, yes. Just do an NVM backup in Z-Wave JS UI, and restore that on the new stick.
Do you mean replacing old stick with a new one same model ? Or do I change brand/model and do backup and restore ? 

> If you're located in EU, you should know though that the Z-Stick 7 has severe RF range issues for an unknown reason.
Never hear that ... good to know.

> I've had the best performance with 500 series controllers and those 700 series controllers that you put on the GPIO of a Raspberry Pi (or similar).
I'm running docker image on a QNAP , and planning to use an Aeotec ZStick7.
So I think that is that a model that I need to be worried about.

Meanwhile, exclude and re-include does the trick ..... don't know why.
[image]

Here it is the log, in case can be helpfull
[zwavejs_2023-08-17.zip](https://github.com/zwave-js/node-zwave-js/files/12370371/zwavejs_2023-08-17.zip)

closed 2023-08-17 by candrea77