zwave-js/zwave-js #6184: CT100 Thermostat Showing as two climate entities, and no fan

txwindsurfer, 2023-08-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? See https://github.com/zwave-js/node-zwave-js/issues/6017.  The CT100 has the same issue that was just fixed on the CT101.  I have 3 CT101 thermostats and all are now fixed however my 1 CT100 still exhibits the same behavior as the CT101s did before the recent update.

What do you observe?  Same outcome as was reported in https://github.com/zwave-js/node-zwave-js/issues/6017

What did you expect to happen?  Only 1 thermostat and a fax

Steps to reproduce the behavior:  Re-interview node
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer:  Radio Thermostat of America
Model name:  CT100
Node ID in your network: 9 

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 2023-08-02
`node-zwave-js` branch: stable
`zwave-js-ui` branch: stable

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[home-assistant_zwave_js_2023-08-17T16-11-37.467Z.log](https://github.com/zwave-js/node-zwave-js/files/12371497/home-assistant_zwave_js_2023-08-17T16-11-37.467Z.log)

txwindsurfer, 2023-08-17:
Opps, new log file uploaded

closed 2023-08-18 by zwave-js-bot

MinchinWeb, 2023-11-04:
Came across this problem, even though I'd updated ZWave-JS (plus, it was showing a Radon sensor??). Solution was to **Re-interview the node**, and then delete the "extra" entries in Home Assistant.

(Hopefully this helps someone else...)

AlCalzone, 2023-11-05:
> plus, it was showing a Radon sensor??

https://zwave-js.github.io/node-zwave-js/#/troubleshooting/nonsensical-values

PR #6185 (2023-08-18): fix(config): remove unnecessary endpoint functionality for CT100
fixes: #6184