zwave-js/zwave-js #6193: Problem with Eurotronics Spirit ZWAVE+ Thermostat and Fibaro FGWPE/F Wall Plug Gen5+

harrykausl, 2023-08-21:
Hello,
I switched in Domoticz from OZWave to zwave-js-ui. I installed it via snap on my pi4. My sensative spirit window sensors work well, my vision security ZG8101 Garage Door Detector too, My Eurotronics spirit zwave+ Thermostat is not recognised. It shows unknown in the manufacturer. No devices are created. The manufacturere for the Fibaro wall plug is unknown too, but there are some devices createt. I am missing the device to switch the plug on or off. I reintervied the 2 devices with debug log on. Here's the result. Perhaps someone can help me.

PS: In OZWave they work without problems.

2023-08-18 09:47:33.640 DEBUG SOCKET: New connection 8bi29UFz7abWwBUcAAAD
2023-08-18 09:47:34.782 DEBUG SOCKET: Event INITED emitted to 8bi29UFz7abWwBUcAAAD
2023-08-18 09:47:34.817 INFO APP: GET /api/settings 304 1231.359 ms - -
2023-08-18 09:47:34.831 INFO APP: GET /api/auth-enabled 304 5.229 ms - -
2023-08-18 09:48:17.358 DEBUG MQTT: Publishing to zwave/Thermostat_Bad/lastActive: { time: 1692344897357, value: 1692344897357 } with options { qos: 1, retain: true }
2023-08-18 09:48:17.608 DEBUG MQTT: Publishing to zwave/Thermostat_Bad/lastActive: { time: 1692344897608, value: 1692344897554 } with options { qos: 1, retain: true }
2023-08-18 09:48:20.848 DEBUG MQTT: Publishing to zwave/Steckdose/lastActive: { time: 1692344900848, value: 1692344900848 } with options { qos: 1, retain: true }
2023-08-18 09:48:20.862 INFO Z-WAVE: [Node 018] Metadata updated: 49-0-Power
2023-08-18 09:48:20.867 DEBUG MQTT: Publishing to zwave/Steckdose/49/0/Power: {
time: 1692344900865,
value: 0,
nodeName: 'Steckdose',
nodeLocation: ''
} with options { qos: 1, retain: true }
2023-08-18 09:48:20.872 INFO Z-WAVE: [Node 018] Value updated: 49-0-Power 0 => 0
2023-08-18 09:48:21.099 DEBUG MQTT: Publishing to zwave/Steckdose/lastActive: { time: 1692344901098, value: 1692344900848 } with options { qos: 1, reta

harrykausl, 2023-08-28:
[zwavejs_2023-08-28.log](https://github.com/zwave-js/node-zwave-js/files/12456514/zwavejs_2023-08-28.log)

AlCalzone, 2023-08-29:
If one of those devices was included using Security S2 or S0, Z-Wave JS needs to have the original keys used to include them. As OZW doesn't support S2, I guess they were included using Security S0 and you didn't copy the original S0 key ("network key") into the Z-Wave JS configuration.

Without this, you cannot fully communicate with the devices.

harrykausl, 2023-08-29:
I copied the network key to S2 ind zwave-js-ui zwave config. In domoticz there is only one key. So I made a s0-key in zwave-js-ui via the config-page. Or have I to use the Domoticz network key for both, s0 and s2? Thanks.

harrykausl, 2023-08-30:
Now I changed the keys, the Eurotronic Spirit is recogniced, but only with one device in Domoticz (air temperature). The plug isn't recogniced. I think, I'll have to delete the 2 devices in zwavejsui and reinclude them.

AlCalzone, 2023-08-30:
Re-including is rarely necessary. Did you try checking the "reset security classes" checkbox when re-interviewing the plug?

harrykausl, 2023-08-30:
The trick helped. My only persisting problem is, I am missing the mode device to switch the mode for the thermostat. In domoticz I set mode 0 (off), 1 (heating) and 3 (turbo). I attached a list of my created devices. Thanks alot. 
[spirit.pdf](https://github.com/zwave-js/node-zwave-js/files/12474425/spirit.pdf)

AlCalzone, 2023-08-30:
Debugging that will require another driver log of a re-interview. No need to check the "reset security" checkbox this time.

harrykausl, 2023-08-30:
Works fine now, thanks a lot!

closed 2023-08-30 by harrykausl