zwave-js/zwave-js #6194: Driver: Failed to open the serial port: Error: Not a tty setting custom baud rate of 115200 (ZW0100)

Eagleman7, 2023-08-21:
I am currently migrating from RHEL 8 to Ubuntu 22.04. I have the Aeotec USB Z-stick 7 attached to a VM on vSphere. I can read the information from the USB device:

```
root@k3s-staging-1:/home/ansible# lsusb -d 10c4:ea60 -v

Bus 001 Device 004: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x10c4 Silicon Labs
  idProduct          0xea60 CP210x UART Bridge
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 CP2102N USB to UART Bridge Controller
  iSerial                 3 40807fee5284e911a58b4992d6f3a178
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

The device is then forwarded to a k3s pod. I can see the device inside the container. However I cannot query it (this is the same behavior on RHEL 8 where it does work).

```
/usr/src/app # lsusb -d 10c4:ea60 -v

> Bus 001 Device 004: ID 10c4:ea60
Bus 001 Device 001: ID 1d6b:0001
Bus 001 Device 002: ID 0e0f:0003
Bus 002 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 0e0f:0002
```

In the front-end I see the following error:
[image]

I cannot find anyone else with the same error in zwave-js-ui. So I'm a bit lost what else I can check.

AlCalzone, 2023-08-21:
What exactly did you configure as the port in the settings?

Eagleman7, 2023-08-22:
Hi AlCalzone,

I configured my OS udev to set it to a fixed value:

/etc/udev/rules.d/99-usb-serial.rules
`SUBSYSTEM=="usb", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", SYMLINK+="ttyUSB-ZStick-5G"`

Then I assigned it in the container like this:
```
  values:
    image:
      repository: ghcr.io/zwave-js/zwave-js-ui
      tag: 8.23.0@sha256:573869081f6e0a2cc95ee086ccda26d41f49278539692e4fac1ddfe932440aea
    ...
      usb:
        enabled: true
        type: hostPath
        hostPath: /dev/ttyUSB-ZStick-5G
        hostPathType: CharDevice
```

It showed the device inside the container:
/usr/src/app # lsusb -d 10c4:ea60 -v
```

> Bus 001 Device 004: ID 10c4:ea60
Bus 001 Device 001: ID 1d6b:0001
Bus 001 Device 002: ID 0e0f:0003
Bus 002 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 0e0f:0002
```

I couldn't be bothered to troubleshoot this further, I already spend too much time on it. I migrated the container to my TrueNAS Scale system where the forwarded does work. In the future I will migrate thius to a Raspberry PI device so it can be fully outside my cluster, and in a more central place in my home.

AlCalzone, 2023-08-22:
If you want to get away from USB forwarding, using TCP may be an alternative:
https://zwave-js.github.io/node-zwave-js/#/usage/tcp-connection

closed 2023-09-09 by zwave-js-assistant[bot]