zwave-js/zwave-js #6195: Message Dead then alive only on all my FGR223

WorldOfGZ, 2023-08-21:
Hello,

I have a strange situation with my Fibaro FGR223. I have 8 modules with all same firmware v5.1
With automation I launch all of them with a delay in Up/Down or positionning.

Every time I have a message on one-two of them (never the same) that the module is dead, and seconds after module is alive.

I upgraded in 8.23.0. It seems appear less than in 8.22.0 but I still have at least 4 messages by days. 

This is only on this module, not on all other modules or brands. Do you know if there a special thing I can do on these FGR223 ?
Thanks

AlCalzone, 2023-08-22:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

The log should include one of the problematic interactions.

WorldOfGZ, 2023-08-22:
@AlCalzone No problem, I will do it.
Just to be sure before, dit I let the manual route at 40kbps or should I have to go back to automatic route ?
Thanks

AlCalzone, 2023-08-22:
Doesn't really matter, as long as the problem happens in the logfile.

WorldOfGZ, 2023-08-22:
Good Evening @AlCalzone 

Please find the log for the automation of this evening.
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12412293/zwave-js-ui-store.zip)

I received messages for node 79
Dead at 19:23:12
Alive at 19:23:14

Hope it can help. Thanks

AlCalzone, 2023-08-23:
This has the typical signs of bad signal strength:
- Most RSSI measurements on the controller are around -90 dBm, which is pretty close to background noise. Some repeaters report much better values (around -60/-70), so this seems to be an issue with the controller range.
- Lots of incoming commands are repeated, again hinting at a suboptimal controller range. This happens when devices don't get the acknowledgement that their commands were received.
- Some outgoing commands fail with status `transmit status: Fail, took 0 ms`, which means the controller is jammed and cannot transmit (due to the above)
- Only around 30 of the ~600 outgoing commands succeeded with 100 kbit/s, most need 40 or even 9.6 kbit/s. Several commands have delays upwards of 1-10 seconds.

Work through these links if you haven't already:
1. https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
2. If there are still issues afterwards: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength
3. Always a good idea: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=optimizing-the-reporting-configuration

Also, which Z-Wave stick do you have?

WorldOfGZ, 2023-08-23:
Thanks for your time. 
My stick is a Zooz 700 Series Z-Wave Plus

Currently it is in direct connection to the Nuc. I will try with a extension cord and follow others tips if it's not solve.
I was surprized that I had problems only on FGR223

AlCalzone, 2023-08-23:
> I was surprized that I had problems only on FGR223

You are having problems with other devices too, just not as severe.

WorldOfGZ, 2023-08-28:
Thanks it’s working better now. 
sorry, last question !
It is considered as normal that all my motion sensor Fibaro FGMR-001 have in graph a speed at 9kbps, when there is a jump ?
When no jump link is green, but when there is a jump the color is brown even if the node in middle is green with the controler.

AlCalzone, 2023-08-28:
That means the connection between the repeater and the target node is **likely** bad.

Or this is now outdated information, but the controller keeps using the slow route until there's a need to change.
You can try doing a single node heal for the problematic ones, hoping it will result in a better route.

closed 2023-08-31 by WorldOfGZ