zwave-js/zwave-js #6196: FGR-223 inclusion fails with `TypeError: Cannot convert undefined or null to object`

maxenced, 2023-08-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Inclusion of FGR-223

What do you observe?

Trying to include two FGR-223 shutter module. They both start inclusion, ask for pin, but are then stuck at "OverwriteConfig". I use latest zwave js version. I've tried to re-interview them. Everything seems good except last part:

```
17:56:24.521 DRIVER   Dropping message because it could not be deserialized: The command class Appli
                      cation Status is not implemented (ZW0303)
17:56:24.522 SERIAL » [ACK]                                                                   (0x06)
17:56:24.996 CNTRLR   [Node 056] [translateValueEvent: value added]
                        commandClass: Central Scene
                        endpoint:     0
                        property:     interviewComplete
                        propertyKey:  undefined
                        internal:     true
                        secret:       false
                        event source: undefined
17:56:25.001 CNTRLR   [Node 056] [+] [Central Scene] interviewComplete: true [Endpoint 0] [internal]
2023-08-21 17:56:25.004 INFO Z-WAVE: [Node 056] Interview stage COMMANDCLASSES completed
17:56:25.008 CNTRLR   [Node 056] Interview stage completed: CommandClasses
2023-08-21 17:56:25.013 INFO Z-WAVE: [Node 056] Interview stage OVERWRITECONFIG completed
17:56:25.016 CNTRLR   [Node 056] Interview stage completed: OverwriteConfig
2023-08-21 17:56:25.032 ERROR APP: Unhandled Rejection, reason: TypeError: Cannot convert undefined or null to object
17:56:25.104 DRIVER   all queues idle
```

I attach the node's json. Firmware version is 5.1

### Device information

Manufacturer: Fibaro
Model name: FGR-223
Node ID in your network: 56

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

`zwave-js-ui` branch: latest

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2023-08-21.log](https://github.com/zwave-js/node-zwave-js/files/12399495/zwavejs_2023-08-21.log)

Node's json : 
```json
{
  "id": 56,
  "name": "",
  "loc": "",
  "values": [],
  "groups": [],
  "neighbors": [],
  "ready": false,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": false,
  "eventsQueue": [
    {
      "time": "2023-08-21T17:48:01.946Z",
      "event": "alive",
      "args": [
        0
      ]
    }
  ],
  "status": "Alive",
  "interviewStage": "OverwriteConfig",
  "priorityReturnRoute": {},
  "customReturnRoute": {},
  "customSUCReturnRoutes": [],
  "applicationRoute": false,
  "hexId": "0x010f-0x0303-0x1000",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x010f:0x0303:0x1000:0.0",
  "manufacturerId": 271,
  "productId": 4096,
  "productType": 771,
  "deviceConfig": {
    "filename": "/usr/src/app/store/.config-db/devices/0x010f/fgr223.json",
    "isEmbedded": true,
    "manufacturer": "Fibargroup",
    "manufacturerId": 271,
    "label": "FGR223",
    "description": "Roller Shutter 3",
    "devices": [
      {
        "productType": 771,
        "productId": 4096
      },
      {
        "productType": 771,
        "productId": 12288
      },
      {
        "productType": 771,
        "productId": 16384
      }
    ],
    "firmwareVersion": {
      "min": "0.0",
      "max": "255.255"
    },
    "preferred": false,
    "paramInformation": {
      "_map": {}
    },
    "proprietary": {
      "fibaroCCs": [
        38
      ]
    }
  },
  "productLabel": "FGR223",
  "productDescription": "Roller Shutter 3",
  "manufacturer": "Fibargroup",
  "firmwareVersion": "5.1",
  "protocolVersion": 3,
  "zwavePlusVersion": 1,
  "zwavePlusNodeType": 0,
  "zwavePlusRoleType": 5,
  "nodeType": 1,
  "endpointsCount": 2,
  "endpoints": [
    {
      "index": 0,
      "label": "Root Endpoint"
    },
    {
      "index": 1,
      "label": "Endpoint 1"
    },
    {
      "index": 2,
      "label": "Endpoint 2"
    }
  ],
  "isSecure": true,
  "security": "S2_Authenticated",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 17,
    "specific": 6
  },
  "lastActive": 1692640081943,
  "firmwareCapabilities": {
    "firmwareUpgradable": false
  },
  "deviceId": "271-4096-771",
  "statistics": {
    "commandsTX": 1,
    "commandsRX": 0,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0,
    "rtt": 55.1,
    "lastSeen": "2023-08-21T17:48:01.943Z"
  },
  "_name": "NodeID_56",
  "prioritySUCReturnRoute": false
}
```

maxenced, 2023-08-22:
Interesting fact : it worked downgrading to 8.15

AlCalzone, 2023-08-22:
I'm able to reproduce. Fix is on the way.

closed 2023-08-22 by AlCalzone

AlCalzone, 2023-08-22:
Z-Wave JS UI v8.23.1 is releasing now.

Arminooo, 2023-08-22:
Just for your information, the issue is/was not only for FGR-223, but also at the FGR-222 shutters. (Fibaro Shutter 2 [FGR-222] and Fibaro Shutter 3 [FGR-223].

AlCalzone, 2023-08-23:
Same root issue, same solution