zwave-js/zwave-js #6197: OTA Firmware update gets stuck after transmitting a fragment fails

AlCalzone, 2023-08-21:
```
2023-08-20T15:30:54.641Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 83
                                    │ security class:  S2_Authenticated
                                    └─[FirmwareUpdateMetaDataCCGet]
                                        total # of reports: 1
                                        report number:      428
2023-08-20T15:30:54.641Z CNTRLR » [Node 097] Sending firmware fragment 428 / 5900
2023-08-20T15:30:54.643Z DRIVER   all queues busy
2023-08-20T15:30:54.644Z SERIAL » 0x013c00a9000100612e9f037f00e5987a9084fd488daaa02a19f15957acb6fd165 (62 bytes)
                                  c361e4afde63b0cfd5acf8d729dd42e018e3f1e103e9f25000000000e0d
2023-08-20T15:30:54.644Z DRIVER » [Node 097] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      14
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 127
                                    │ security class:  S2_Authenticated
                                    └─[FirmwareUpdateMetaDataCCReport]
                                        report #: 428
                                        is last:  false
2023-08-20T15:30:54.654Z SERIAL « [ACK]                                                                   (0x06)
2023-08-20T15:30:54.655Z SERIAL « 0x010401a90152                                                       (6 bytes)
2023-08-20T15:30:54.655Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:30:54.656Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2023-08-20T15:30:56.806Z SERIAL « 0x011f00a80000010061119f035400291182b51ab4afaa1aa8ea294900a7007f7fc (33 bytes)
                                  a
2023-08-20T15:30:56.808Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:30:56.809Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 84
                                    │ security class:  S2_Authenticated
                                    └─[FirmwareUpdateMetaDataCCGet]
                                        total # of reports: 1
                                        report number:      429
2023-08-20T15:30:57.449Z SERIAL « 0x011f00a80000010061119f035400291182b51ab4afaa1aa8ea294900a7007f7fc (33 bytes)
                                  a
2023-08-20T15:30:57.451Z DRIVER   Dropping message with invalid payload
2023-08-20T15:30:57.452Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 84)
2023-08-20T15:30:57.452Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:30:58.135Z SERIAL « 0x011f00a80000010061119f035400291182b51ab4afaa1aa8ea294900a7007f7fc (33 bytes)
                                  a
2023-08-20T15:30:58.136Z DRIVER   Dropping message with invalid payload
2023-08-20T15:30:58.136Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 84)
2023-08-20T15:30:58.136Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:31:09.620Z SERIAL « 0x011f00a80000010061119f035400291182b51ab4afaa1aa8ea294900a6007f7fc (33 bytes)
                                  b
2023-08-20T15:31:09.623Z DRIVER   Dropping message with invalid payload
2023-08-20T15:31:09.623Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -90 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 84)
2023-08-20T15:31:09.623Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:31:12.703Z SERIAL « 0x011d00a90e0106ed007f7f7f7f7f000107000000000220b461000000000021    (31 bytes)
2023-08-20T15:31:12.704Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:31:12.705Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            14
                                    transmit status:        NoAck, took 17730 ms
                                    routing attempts:       32
                                    protocol & route speed: Z-Wave, 40 kbit/s
[... 76 lines cut ...]
                                        total # of reports: 1
                                        report number:      429
2023-08-20T15:31:47.763Z DRIVER   all queues busy
2023-08-20T15:31:47.765Z SERIAL » 0x0103003bc7                                                         (5 bytes)
2023-08-20T15:31:47.765Z DRIVER » [REQ] [GetBackgroundRSSI]
2023-08-20T15:31:47.770Z SERIAL « [ACK]                                                                   (0x06)
2023-08-20T15:31:47.771Z SERIAL « 0x0107013b9093937f2d                                                 (9 bytes)
2023-08-20T15:31:47.771Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:31:47.772Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -112 dBm
                                    channel 1: -109 dBm
                                    channel 2: -109 dBm
2023-08-20T15:31:47.774Z DRIVER   all queues idle
2023-08-20T15:31:56.785Z SERIAL « 0x011f00a80000010061119f0358009756a138f98391cd15d0eb8b1100a7007f7f9 (33 bytes)
                                  0
2023-08-20T15:31:56.788Z SERIAL » [ACK]                                                                   (0x06)
2023-08-20T15:31:56.789Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 88
                                    │ security class:  S2_Authenticated
                                    └─[FirmwareUpdateMetaDataCCGet]
                                        total # of reports: 1
                                        report number:      429
```

niemyjski, 2023-08-22:
I'm also seeing this when applying the aeotec 1.5 firmware. It just stops randomly.. seems like the controller should retry and log it out and or cancel the upgrade if no health check.

ocalvo, 2023-08-23:
I am seeing something similar.

The upgrade has been hung since 24 hours ago when I started.

sikksakk, 2023-08-24:
i see the same, seems to happen more often on devices with poorer connection.
using aeotec z-stick gen7 and currently updating multisensor 7's

AlCalzone, 2023-08-24:
Thanks for the feedback, but no need to comment further, as this just adds additional noise.

GeertvanHorrik, 2023-08-29:
I encountered this as well. Using ZWave PC controller, I was able to reliably update a node that was continuously failing via zwave-js. I checked the [code](https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/node/Node.ts#L5076) and the logs from the zwave controller. Maybe it's because a node can request the same fragment multiple times? Here is (some) logging, where I think the red part is the request from the node:

[image]

Could the bug be related to the "premature request" storage? If the premature request is not stored, the node will re-request the fragment if I understand the logs correctly?

AlCalzone, 2023-08-29:
As soon as you can update to an application version that includes https://github.com/zwave-js/node-zwave-js/releases/tag/v11.13.1, please re-test.
I fixed a bug where some commands were getting sent but this wasn't visible to the API used to send them, essentially causing Z-Wave JS to get stuck. This might be the root cause for this aswell.

Looks like the HA core addon is already up to date: https://zwave-js.github.io/which-version/

GeertvanHorrik, 2023-08-29:
Thanks for the reply. 

I am on latest stable HA (2023.8.4), but zwave js is version 11.13.0. I assume it will be included in the next (September, 2023.09) release. Will test as soon as I have the new release.

GeertvanHorrik, 2023-08-29:
Thanks for the clarification, that makes sense. I am using Z-Wave JS UI. I was confused because I have disabled the _MQTT Gateway (Enable this to use Z-Wave JS UI only as Control Panel)_. 

When hovering the (i) symbol in Z-Wave JS UI, I can see the versions now:

[image]

Will await the update.

GeertvanHorrik, 2023-09-06:
Just received the update via Home Assistant. Immediately tested and the first firmware update succeeded!

Thank you very much for the great continued work!

AlCalzone, 2023-09-06:
Thanks for reporting back. I'll wait and see if this is fixed for other users too before closing this issue.

elreydetodo, 2023-09-06:
I was also able to do two firmware updates after updating z-wave-js ui addon today. These updates had been repeatedly failing over the past week.

abscondment, 2023-09-06:
I'm having repeated failure updating an aeotec ZWA039, even with 11.14.0. I have been able to get it stuck with an update pending forever (not indicated in the UI, but can't start a new update or soft reset the controller).

I'm not certain this is triggered by a failed fragment or even the same symptoms experienced here. Maybe a different issue?

mikedahlgren, 2023-09-06:
I had one node that I had attempted to do a firmware update 5x times and each time it would get stuck at a different point.  After updating to 11.14.0 it worked perfectly!  Awesome work!

AlCalzone, 2023-09-07:
@abscondment 
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of the update attempt, loglevel `debug` and attach it here as a file (drag & drop into the text field).

AlCalzone, 2023-09-08:
Any specific device exhibiting that behavior?

abscondment, 2023-09-09:
> Any specific device exhibiting that behavior?

Yes, an Aeotec [ZWA039 to 2.0.3](https://aeotec.freshdesk.com/support/solutions/articles/6000256075-update-zwa039-aerq-sensor-v2-03) (called [2.3.0 in the repo](https://github.com/zwave-js/firmware-updates/blob/main/firmwares/aeotec/ZWA039-A.json)? Link/shasum, are correct).

AlCalzone, 2023-09-25:
I'll treat the lack of further reports as a sign that this is fixed. If not, please open a new issue.

closed 2023-09-25 by AlCalzone