zwave-js/zwave-js #6199: Almost all nodes unknown or dead after updating node-zwave-js 11.12.0 (via upgrade to zwavejsui 8.23.0)

ridizy, 2023-08-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I upgraded my docker container for zwavejsui today to 8.23.0.  I noticed many of my devices eventually went into an unknown state.  I tried rolling back to various versions as far back as 8.22.0

What do you observe?
I noticed many of my devices eventually went into an `unknown` state.  I tried rolling back to various versions as far back as 8.22.0 and on some versions the nodes are marked `dead` instead of `unknown`

What did you expect to happen? 
I expected to update to complete and my z-wave network to continue to function as it has before

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.23.0.73606d4
zwave-js: 11.12.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

A little while after updating zwavejsui, I did an OTW update of Inovelli light switch.  That succeeded and I went to do the next one when I noticed the issue.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

This z-wave network has been running without issues for a few months.  I usually update my containers every few weeks.

### Attach Driver Logfile

[zwavejs_2023-08-21.log](https://github.com/zwave-js/node-zwave-js/files/12402384/zwavejs_2023-08-21.log)

gsemet, 2023-08-21:
Pretty similarly, all my mains powered devices seemed to have being disconfigured, still appearing but with no configuration (vundefined).

AlCalzone, 2023-08-22:
I'll need to see the driverlog on loglevel "debug" to know for sure, but this looks like a combination of the stick acting up/becoming unresponsive and some node flooding the network.

As a quick troubleshoot, try unplugging and re-plugging the stick.

gsemet, 2023-08-22:
I had to cold restart the server 3 times until it went back to normal.

ridizy, 2023-08-22:
Thanks for the tips.  I did try unplugging and re-plugging the stick and also a cold restart.  Neither seemed to help.  

Here is the the driver log with it correctly set to "debug" this time.  
[zwavejs_2023-08-22.log](https://github.com/zwave-js/node-zwave-js/files/12407974/zwavejs_2023-08-22.log)

As a troubleshooting step, I did try to re-interview Node 4 yesterday and that seems to still be having an issue?

AlCalzone, 2023-08-22:
@ridizy Looks like there is some problem trying to reach node 4. Pinging it fails after 27 seconds (!) with a status that normally indicates that the controller is jammed by too much traffic (which isn't the case here).

The difference in the newest version is that Z-Wave JS thinks this is a temporary issue with the controller and keeps retrying, as opposed to marking the node as dead.

Any clue what's going on with node 4 that could cause this? Maybe that one needs to be power cycled.

ridizy, 2023-08-22:
Thanks @AlCalzone.  It was one of the first ones marked as "unknown" yesterday, so I tried to re-interview it (not sure if that was smart).  

It's a light switch and I've tried pulling the air gap on it a couple times.  Can I mark it "dead" somehow to get passed the issue and then come back to troubleshoot it later?

AlCalzone, 2023-08-22:
I'm afraid you'll have to go back to v8.22.3 until I've fixed this.

ridizy, 2023-08-22:
Thank you @AlCalzone.  I've rolled back to v8.22.3 and it's not getting hung on node 4 and I was able to remove it.  However, it is marking most of the rest, but not all, of the nodes dead.  

I also tried a cold reboot of the host, unplugging and re-plugging the stick and a network heal.

Here is an updated driver log file: 
[zwavejs_2023-08-22.log](https://github.com/zwave-js/node-zwave-js/files/12409722/zwavejs_2023-08-22.log)

AlCalzone, 2023-08-22:
The ping to node 2 goes fine, but all the others fail with the same problem as node 4 now.

Just to exclude a software issue, you could go back all the way to v8.21.2, which is the version before I switched node IDs used in commands from 8 to 16 bit. Although if your controller had a problem with that, the ping to node 2 shouldn't work either.

If that's not it, maybe all your problematic nodes are routing through a single point of failure.

Did you follow these recommendations?
https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

AlCalzone, 2023-08-22:
I'll keep the issue open because there's still the infinite loop in this case I need to address.

PR #6211 (2023-08-24): fix: only consider controller as jammed on status Fail with 0 TX ticks
fixes: #6199
closed 2023-08-24 by zwave-js-bot