zwave-js/zwave-js #6208: No configuration options for newly joined devices

MRobi1, 2023-08-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Included 3 "new" devices that were previously on the network

What do you observe?
Inclusion completes fine. Able to control on/off of devices without issue. However no configuration options are present.

What did you expect to happen?
Configuration options for devices to show up

Steps to reproduce the behavior:
Include new device to network

Newly included device:
[image]

Same model device already on network:
[image]

### Device information

Manufacturer: Inovelli
Model name: LZW31-SN and LZW30-SN
Node ID in your network: 76, 77, 78

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.23.0
zwave-js: 11.12.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Moved some devices to new positions in house over the period of about a week.
- Excluded devices from network and uninstalled
- A few days later re-installed in new location and included devices

### Did this work before?

Yes (please describe)

### If yes, where did it work?

These 3 devices were all previously on the network. Also have several of the same model devices already on the network and configuration options are all there.

Have attempted to re-interview several times. Tried refresh values/update topics. Factory reset devices. Exclude/Include multiple times. Restart ZwaveJS. Restart HA. Restart server. Unplug/re-plug zwave stick. Have tried to manually update parameters through HA which throws an error.

### Attach Driver Logfile

[zwavejs_2023-08-23.log](https://github.com/zwave-js/node-zwave-js/files/12422694/zwavejs_2023-08-23.log)

AlCalzone, 2023-08-24:
```
2023-08-23T19:28:28.577Z CNTRLR   [Node 076] User-provided device config loaded
2023-08-23T19:28:28.580Z CNTRLR   [Node 077] User-provided device config loaded
2023-08-23T19:28:28.584Z CNTRLR   [Node 078] User-provided device config loaded
```
You're not using the official config files for those.

MRobi1, 2023-08-24:
> You're not using the official config files for those.
Well that's interesting. How do I change that?

AlCalzone, 2023-08-25:
> As for why the custom config files worked for probably almost 2 years, and also on all the other devices is beyond me

The format for config parameters changed long ago from an object to an array. A while ago I had to remove the code that parses the legacy format. Your config files were probably still in the old format, and this broke after re-interviewing.