zwave-js/zwave-js #6213: Missing device configuration: Shelly Qubino wave 1PM

nberlee, 2023-08-24:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Shelly Qubino Wave 1PM

### Manufacturer ID

0x0460

### Product Type

0x0002

### Product ID

0x0084

### Firmware Version

FW: v10.13.0 SDK: v7.19.1

### Do you have a manual?

https://kb.shelly.cloud/knowledge-base/wave-1pm

closed 2023-08-24 by nberlee

AlCalzone, 2023-08-25:
Why close?

AlCalzone, 2023-08-25:
oh, yeah