zwave-js/zwave-js #6214: Missing device configuration: Shelly Qubino Wave 2PM

nberlee, 2023-08-24:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Shelly Qubino Wave 2PM

### Manufacturer ID

0x0460

### Product Type

0x0002

### Product ID

0x0081

### Firmware Version

FW: v10.27.0 SDK: v7.19.1

### Do you have a manual?

https://kb.shelly.cloud/knowledge-base/wave-2pm

lieblinger, 2023-09-26:
@QubinoHelp Could you help us with the integration into ZWave-JS?

QubinoHelp, 2023-09-26:
@lieblinger 
We're working on it, Wave 1PM is up, the Wave 1 and 2PM is work in progress. Most probably everything will be ready for the end of this week.

nberlee, 2023-09-29:
As PR #6326 is merged this issue may be closed as complete

closed 2023-09-29 by nberlee