zwave-js/zwave-js #6215: Heatit Thermostat Z-TRM3 lost Multilevel Sensor capability

sjarvela, 2023-08-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Since updating to latest version (11.13.0), all my Heatit Z-TRM3 thermostats stopped reporting multilevel sensor values, ie thermostat floor and air temperatures. Log is showing "[Node nnn] does not support CC Multilevel Sensor - discarding received command", which seems to relate to change #6178.

All thermostats had multilevel sensor shown in node, they were just not updating due to this change. Now that I re-interviewed two of these thermostats, multilevel sensor is lost in UI as well.

What might be the reason zwave-js now thinks these devices don't support this? Attached driver log has interview of node 024, and for me it seems to say it does, for example

```
2023-08-25T07:42:48.693Z DRIVER « [Node 024] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 81
                                    │ security class:  S2_Authenticated
                                    └─[VersionCCCommandClassReport]
                                        CC:      Multilevel Sensor
                                        version: 5
2023-08-25T07:42:48.694Z CNTRLR   [Node 024]   supports CC Multilevel Sensor (0x31) in version 5    [Endpoint 2]
```

In version 11.6.0 there was no problems, and with another zwave-js instance running with older version, Z-TRM3 still operates without problems.

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 11.13.0

### Did you change anything?

yes (please describe)

### Did this work before?

Yes (please describe)

### If yes, where did it work?

In earlier version 11.6.0

### Attach Driver Logfile

[zwave-driver.log](https://github.com/zwave-js/node-zwave-js/files/12437613/zwave-driver.log)

AlCalzone, 2023-08-25:
The endpoint that sends the reports does support Multilevel Sensor CC:
```
2023-08-25T07:42:48.621Z CNTRLR « [Node 024] received secure commands (S2_Authenticated)            [Endpoint 2]
                                  supported CCs:
                                  · Association
                                  · Association Group Information
                                  · Multi Channel Association
                                  · Multilevel Sensor

...

2023-08-25T07:42:48.841Z CNTRLR   [Node 024] does not support CC Multilevel Sensor - discarding received command...
2023-08-25T07:42:48.843Z DRIVER « [Node 024] [REQ] [ApplicationCommand]                              [discarded]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 83
                                    │ security class:  S2_Authenticated
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      2
                                      │ destination: 0
                                      └─[MultilevelSensorCCSupportedSensorReport]
                                          supported sensor types:
                                          · Air temperature
```

markpetrie1980, 2023-08-25:
I'm having the same issue with Z-Wave JS UI and a Fibaro Universal Binary Sensor

CNTRLR does not support CC Multilevel Sensor - discarding received command

tefteulf, 2023-08-26:
Same problem as described above with 2 Fibaro universal binary sensor (2 ch on/off + 1 temp) in Z-Wave JS UI v1.15.7
Downgrade to  Z-Wave JS UI v1.15.6 solves the problem.

closed 2023-08-28 by zwave-js-bot

ScratMan, 2023-08-29:
I updated to ZWave JS UI 8.23.2 to get the fix, but my temperature sensor on the FGK101 immediately stopped updating.
I reverted to 8.23.0 and the temperature has been updated, I'll monitor it to see if it reports correctly.

AlCalzone, 2023-08-29:
8.23.2 doesn't include the fix yet: https://zwave-js.github.io/which-version/

> Latest version:
>     v8.23.2
> Driver version:
>     v11.13.0 

That is in v11.13.1

ScratMan, 2023-08-29:
Oh ! My mistake. I saw a node release before an UI release and thought it was included 😂

quadhammer, 2023-09-26:
I'm getting the same with a Fibaro FGS221.

Logs are being spammed with:

CNTRLR [Node 054] does not support CC Meter - discarding received command...

Here are the logs on an include:

`023-09-26T03:46:30.337Z CNTRLR [Node 054] Beginning interview - last completed stage: None
2023-09-26T03:46:30.337Z CNTRLR [Node 054] new node, doing a full interview...
2023-09-26 13:46:30.338 INFO Z-WAVE: [Node 054] Interview started
2023-09-26T03:46:30.338Z CNTRLR » [Node 054] querying protocol info...
2023-09-26T03:46:30.372Z CNTRLR « [Node 054] received response for protocol info:
basic device class: Routing Slave
generic device class: Binary Switch
specific device class: Binary Power Switch
node type: End Node
is always listening: true
is frequent listening: false
can route messages: true
supports security: false
supports beaming: true
maximum data rate: 40000 kbps
protocol version: 3
2023-09-26 13:46:30.373 INFO Z-WAVE: [Node 054] Interview stage PROTOCOLINFO completed
2023-09-26T03:46:30.374Z CNTRLR [Node 054] Interview stage completed: ProtocolInfo
2023-09-26T03:46:30.375Z CNTRLR » [Node 054] querying node info...
2023-09-26T03:46:30.530Z CNTRLR « [Node 054] node info received
supported CCs:
· No Operation
· No Operation
· Z-Wave Protocol
· No Operation
· No Operation
· No Operation
· No Operation
· No Operation
· No Operation
· Z-Wave Protocol
· No Operation
· No Operation
2023-09-26T03:46:30.531Z CNTRLR « [Node 054] node info received
supported CCs:
· No Operation
· No Operation
· Z-Wave Protocol
· No Operation
· No Operation
· No Operation
· No Operation
· No Operation
· No Operation
· Z-Wave Protocol
· No Operation
· No Operation
2023-09-26 13:46:30.532 INFO Z-WAVE: [Node 054] Interview stage NODEINFO completed
2023-09-26T03:46:30.532Z CNTRLR [Node 054] Interview stage completed: NodeInfo
2023-09-26T03:46:30.533Z CNTRLR [Node 054] Interviewing Binary Switch...
2023-09-26T03:46:30.533Z CNTRLR » [Node 054] querying Binary Switch state...
2023-09-26 13:46:30.729 INFO Z-WAVE: [Node 054] Metadata updated: 37-0-currentValue
2023-09-26 13:46:30.731 INFO Z-WAVE: [Node 054] Value added: 37-0-currentValue => false
2023-09-26 13:46:30.731 INFO Z-WAVE: [Node 054] Metadata updated: 37-0-targetValue
2023-09-26 13:46:30.732 INFO Z-WAVE: [Node 054] Metadata updated: 37-0-duration
2023-09-26T03:46:30.732Z CNTRLR « [Node 054] received Binary Switch state:
current value: false
2023-09-26 13:46:30.732 INFO Z-WAVE: [Node 054] Interview stage COMMANDCLASSES completed
2023-09-26T03:46:30.733Z CNTRLR [Node 054] Interview stage completed: CommandClasses
2023-09-26 13:46:30.733 INFO Z-WAVE: [Node 054] Interview stage OVERWRITECONFIG completed
2023-09-26T03:46:30.733Z CNTRLR [Node 054] Interview stage completed: OverwriteConfig
2023-09-26 13:46:30.733 INFO Z-WAVE: [Node 054] Interview stage COMPLETE completed
[... 53 lines cut ...]
· Z-Wave Protocol
· No Operation
· No Operation
2023-09-26 13:47:04.389 INFO Z-WAVE: [Node 054] Interview stage NODEINFO completed
2023-09-26T03:47:04.391Z CNTRLR [Node 054] Interview stage completed: NodeInfo
2023-09-26T03:47:04.392Z CNTRLR [Node 054] Interviewing Binary Switch...
2023-09-26T03:47:04.393Z CNTRLR » [Node 054] querying Binary Switch state...
2023-09-26 13:47:04.557 INFO Z-WAVE: [Node 054] Metadata updated: 37-0-currentValue
2023-09-26 13:47:04.559 INFO Z-WAVE: [Node 054] Value added: 37-0-currentValue => false
2023-09-26 13:47:04.560 INFO Z-WAVE: [Node 054] Metadata updated: 37-0-targetValue
2023-09-26 13:47:04.560 INFO Z-WAVE: [Node 054] Metadata updated: 37-0-duration
2023-09-26T03:47:04.561Z CNTRLR « [Node 054] received Binary Switch state:
current value: false
2023-09-26 13:47:04.561 INFO Z-WAVE: [Node 054] Interview stage COMMANDCLASSES completed
2023-09-26T03:47:04.562Z CNTRLR [Node 054] Interview stage completed: CommandClasses
2023-09-26 13:47:04.562 INFO Z-WAVE: [Node 054] Interview stage OVERWRITECONFIG completed
2023-09-26T03:47:04.563Z CNTRLR [Node 054] Interview stage completed: OverwriteConfig
2023-09-26 13:47:04.563 INFO Z-WAVE: [Node 054] Interview stage COMPLETE completed
2023-09-26T03:47:04.564Z CNTRLR [Node 054] Interview completed
2023-09-26 13:47:04.565 INFO Z-WAVE: [Node 054] Value added 54-37-0-currentValue => false
2023-09-26 13:47:04.565 INFO Z-WAVE: [Node 054] Value added 54-37-0-targetValue => undefined
2023-09-26 13:47:04.565 INFO Z-WAVE: [Node 054] Value added 54-37-0-duration => NaNs
2023-09-26 13:47:04.565 INFO Z-WAVE: [Node 054] Ready: Unknown manufacturer 0xXXXX - Unknown product 0xXXXX (0xXXXX)
2023-09-26T03:47:04.568Z CNTRLR [Node 054] The node is ready to be used
2023-09-26 13:47:04.571 INFO Z-WAVE: [Node 054] Interview COMPLETED, all values are update`

AlCalzone, 2023-09-26:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

That log should include a re-interview.

quadhammer, 2023-09-26:
I turned the driver log to log level 'debug'  before I did my re-interview with the debug log open, but as I am using the built in Z-Wave JS  in home assistant, I can't find where the actual logs are.

AlCalzone, 2023-09-26:
The log you posted above looks a lot like Z-Wave JS UI though? The link I posted also explains how to retrieve the files at the bottom. Make sure to put the driver log on `debug`, the snippet above isn't.

If you're really not using Z-Wave JS UI, check https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs

quadhammer, 2023-09-28:
> If you're really not using Z-Wave JS UI, check https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs

Thank you, that's were it was. :)

[zwave_js(1).log](https://github.com/zwave-js/node-zwave-js/files/12745144/zwave_js.1.log)

quadhammer, 2023-09-28:
I found another debug log that hopefully is of use as well:

```

{
  "id": 54,
  "name": "Bedroom TV",
  "loc": "Bedroom",
  "values": [
    {
      "id": "54-37-0-currentValue",
      "nodeId": 54,
      "toUpdate": false,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "currentValue",
      "propertyName": "currentValue",
      "type": "boolean",
      "readable": true,
      "writeable": false,
      "label": "Current value",
      "stateless": false,
      "commandClassVersion": 2,
      "list": false,
      "value": false,
      "isCurrentValue": true,
      "targetValue": "37-0-targetValue",
      "lastUpdate": 1695878944331,
      "newValue": false
    },
    {
      "id": "54-37-0-duration",
      "nodeId": 54,
      "toUpdate": false,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "duration",
      "propertyName": "duration",
      "type": "duration",
      "readable": true,
      "writeable": false,
      "label": "Remaining duration",
      "stateless": false,
      "commandClassVersion": 2,
      "list": false,
      "value": {
        "value": 41,
        "unit": "minutes"
      },
      "lastUpdate": 1695878944136,
      "newValue": {
        "value": 41,
        "unit": "minutes"
      }
    },
    {
      "id": "54-37-0-targetValue",
      "nodeId": 54,
      "toUpdate": false,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "boolean",
      "readable": true,
      "writeable": true,
      "label": "Target value",
[... 190 lines cut ...]
  "statistics": {
    "commandsTX": 2,
    "commandsRX": 385,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0,
    "rtt": 451.3,
    "lastSeen": "2023-09-28T05:33:43.021Z",
    "rssi": -70,
    "lwr": {
      "protocolDataRate": 2,
      "repeaters": [],
      "rssi": -72,
      "repeaterRSSI": []
    }
  },
  "supportsTime": false,
  "_name": "Bedroom TV (Bedroom)",
  "applicationRoute": false,
  "lastReceive": 1695879223021,
  "errorReceive": false,
  "errorTransmit": false
}

```

AlCalzone, 2023-09-28:
The communication from the device seems completely off - like the capability reports don't make any sense whatsoever.

I'd start by power cycling the thermostat, if that doesn't work power cycle the stick (pull out, put back),
If that doesn't work either, exclude the thermostat, factory reset it, include it again.

quadhammer, 2023-09-29:
sorry, I think I've given you a misdirection. I found this error by googling the same error message, but it's not the same device. It's actually a  Fibaro binary switch, but I'm not sure of the model, because it's in the wall, and it no longer identifies properly: it just gives "unknown device".

Should I open a new issue, despite the same error?

AlCalzone, 2023-09-29:
Doesn't matter which device it is, the troubleshooting steps from my previous post still apply