zwave-js/zwave-js #6229: zwave-JS-UI: Greenwave GWPN1 keeps flashing network error

pa3fat, 2023-08-30:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Snap

### Z-Wave JS UI version

zwave-js-ui: 8.22.3.8607bd8 

### ZwaveJS version

zwave-js: 11.9.2 

### Describe the bug

Last week upgraded to the Zwave-JS-UI to replace the previous Zwave setup supported in Domoticz.

Now i keep getting flashing buttons on the Greenwave GWPN1. I have about 7 of them and all behave alike. Nothing wrong in their functioning on switchting the lights on/off.

Setting the configuration to:
[17-112-0-4] LED for network error
[0] Disable the LED for network error (Default)

does not resolve it.

Is there other setting that resolves this?

### To Reproduce

Tried redefining the settings in the gui but issue remains.
When switch on/off the issue is termporarily gone

### Expected behavior

No blinking lights on the device

AlCalzone, 2023-08-30:
Which firmware version is your device on? Mine has 4.28 and it stops blinking when setting parameter 4 to 0. GET param 4 also works.

pa3fat, 2023-08-30:
Hi ALCalzone,

Thanks for asking.
The devices are on a mix of V4.20/V4.21.

Notice my reported issue is gone in github already, why?

Ron

AlCalzone, 2023-08-30:
The issue was moved to the correct repo (https://github.com/zwave-js/node-zwave-js/issues/6229#issuecomment-1698708553). Just follow the links in that email (prefer responding on Github though, less noisy).

AlCalzone, 2023-08-30:
Hmm, I guess the firmware on yours doesn't have the parameter. Unfortunately with those old devices it's impossible to get a firmware changelog to know when it was added.

It seems like the device expects regular communication from the controller so it doesn't blink the error LED. Pinging seems not enough, but if you poll the binary switch every 2 minutes (or whatever parameter 2 is set to), this keeps it from blinking - and you get notified about manual button presses, although a bit delayed.

I think Z-Wave JS UI has a setting for polling - @robertsLando where was that again?

robertsLando, 2023-08-30:
https://zwave-js.github.io/zwave-js-ui/#/device-support/polling

pa3fat, 2023-08-30:
Implemented as suggested about 30 minutes ago. Checked some and they remain silent now. Polling interval at 30 seconds. Will try to increase gradually and see if this keeps them silent. 

Thanks

AlCalzone, 2023-08-30:
Mine has a configuration parameter (no. 2) which can be used to set that interval. I recommend setting this as high as possible, and then poll just frequently enough. E.g. interval 10 minutes, poll interval 9.5 minutes.

The less traffic on the mesh the better.

pa3fat, 2023-08-30:
Have them on 60minutes.
Checked 2 of them they are identical.

The flashing even with the 30second pull is back. So that did not yet solve it.

pa3fat, 2023-08-30:
[image]

AlCalzone, 2023-08-30:
Please re-interview the plug, make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` of the process and attach it here as a file (drag & drop into the text field).

Your screenshot looks like the device wasn't interviewed completely - the CC versions are missing.
[image]

AlCalzone, 2023-08-30:
I've also tested with mine. Config parameter 1 on 10 minutes, polling frequency (Binary Switch - currentValue) on 540 seconds (9 minutes). 30 minutes later, no flashing.

Two more ideas:
- Other values might need to be polled aswell. I've found Hubitat drivers that also poll the two Meter values, and some even poll the Wheel status (Config param 2)
- The config parameter might have a hidden upper limit, so maybe 60 minutes isn't actually working?

pa3fat, 2023-08-30:
[zwavejs_current(1).log](https://github.com/zwave-js/node-zwave-js/files/12475370/zwavejs_current.1.log)

pa3fat, 2023-08-30:
hope i did it correctly as requested.

pa3fat, 2023-08-30:
Interviewed notes 5, 11,17.
Checked the parameter 4 value which is then again blank. Redefined it at 0 no blinking

will attached current log again

pa3fat, 2023-08-30:
[image]

AlCalzone, 2023-08-30:
It doesn't look like the firmware on your plugs supports parameters 3 and 4. Other than that, the interviews look normal.

Try enabling polling for the two meter values too. Maybe that's all it takes. And if that helps increase the polling time as much as possible to keep the network quiet.

pa3fat, 2023-08-30:
it worked in Domoticz when the previous think it was called OZW was active. Just occasionally had it then. Now all Greenwave devices keep flashing. I have set the 2 meter values also at 500 seconds.

AlCalzone, 2023-08-30:
Any chance you can temporarily use OZW and create a log of what that polls and how frequently?

pa3fat, 2023-08-30:
not sure how to do that without crashing what is working now. Let's give it some time if the pollings show result. They are still blinking....

pa3fat, 2023-08-30:
besides the polling do i need to activate other settings like MQTT as in the Domoticz dashboard the last seen for the lamps are at 01:30 last night. So the polling is not actively updating the status seen by the polling

pa3fat, 2023-08-30:
For now the blinking has stopped. Added the push to MQTT, not sure if this helped as the dashboard in domoticz remains unchanged on the last seen. But it is quiet now. Will try later to remove this MQTT toggle and let you know if this helps. Will now update at least all Greenwave to the 10minute setting so they are alike

pa3fat, 2023-09-03:
Stable offered bypass works.

closed 2023-09-03 by pa3fat