zwave-js/zwave-js #6236: 005107 von Popp & Co, Temperature and Config not working

umtauscher, 2023-09-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
its constant
What do you observe?
the siren is detected, but there are is nothing else detected but an on/off funktion.
The configuration doesn‘t work but instead it links to the correct parameter list https://devices.zwave-js.io/?jumpTo=0x0154:0x0004:0x0002:0.0

There is no way to disable Tamper or read the temperature sensor

What did you expect to happen?

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Popp&Co
Model name: Solar Outdoor siren 
Node ID in your network: 179

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.23.2
zwave-js: 11.13.0

### Did you change anything?

yes (please describe)

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

I am sorry, this is too complicated for me

AlCalzone, 2023-09-04:
Here's how to create the logs in Home Assistant: https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs
And this section describes how to re-interview: https://www.home-assistant.io/integrations/zwave_js#device-panel

To diagnose, please:
1. start logging in a separate window
2. do a re-interview
3. wait until the device is ready again
4. download the log from the log window and drag/drop it here to upload

umtauscher, 2023-09-05:
Thanks for getting back to me and thanks again for the howto produce the proper log.
The NodeID of the siren is 179 btw.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12525012/zwave_js.log)

umtauscher, 2023-09-05:
BTW there are 2 sirens of that brand. One has the temperature sensor (the older one as mine) and one has not.
Cheers

umtauscher, 2023-09-05:
I just realize, that this device is connected with S0 legacy security. The key for this is:
995D23581BE5F2E110898FADB07C1C67

AlCalzone, 2023-09-06:
Is there some kind of filter active in the log? If so, please disable it and retry. I don't see any outgoing commands, only some responses from the stick.

umtauscher, 2023-09-06:
Hmm, I wouldn't know of any filter or how I could set or reset them. Maybe I an doingg something wrong?
I used two different browser instances with HA. One to start the log and the other to reinterview the device. Is that correct?
I have just produced another log in the silly mode, hoping you could get more information from this.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12534727/zwave_js.log)

umtauscher, 2023-09-06:
maybe this is better?
[home-assistant_zwave_js_2023-09-06T07-42-06.026Z.log](https://github.com/zwave-js/node-zwave-js/files/12534801/home-assistant_zwave_js_2023-09-06T07-42-06.026Z.log)

AlCalzone, 2023-09-06:
Hmm no, the first one is still incomplete, the other is unrelated. It looks like you're using Z-Wave JS UI though? In that case there's a better way to get the log:
- Write the logfile: https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs, no need to keep the log tab open
- Download it: https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=download-zip (only the `zwavejs_<date>.log` is required

umtauscher, 2023-09-06:
[zwavejs_2023-09-06.log](https://github.com/zwave-js/node-zwave-js/files/12535992/zwavejs_2023-09-06.log)
There are far too many logging option...
Thanks for your patience

AlCalzone, 2023-09-06:
The device doesn't respond when the list of encrypted commands is requested. Normally this means the security key does not match the one the device has. Try this:
- Exclude the device
- Factory reset it to remove its old key (should be explained in the manual)
- Include it again

umtauscher, 2023-09-06:
Thanks, I will try an will get back to you asap.

umtauscher, 2023-09-06:
Hi Al...
this did the trick. I feel a little bit stupid not to have tried that myselfd before.
thank you very much for your kind help.
Everything works now as expected. Even the configuration and the temperature sensor.
I'm very happ now.
Cheers

closed 2023-09-06 by AlCalzone