zwave-js/zwave-js #6241: Andersen Verilock Translator doesn't provide an entity for locked/unlocked

mt98987987, 2023-09-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe?
The Verilock Translator is a Zwave bridge to Andersen 345Mhz door/window sensors. The 345Mhz sensors replace the window latches or patio door locks and provide open/closed AND locked/unlocked status. They use a small coin cell battery and also provide battery status. I am able to view the 345Mhz sensor transmissions natively using rtl_433 and I can confirm the sensors are functioning correctly and providing the above info. 

The translator is supposed to connect to the zwave network and provide the status of the paired 345Mhz sensors including: open/closed, locked/unlocked, and battery status. An entity is created for each paired sensor for battery status and open/closed and I can see the states change correctly. No entity is created for the locked/unlocked but I see events being fired for lock/unlock. The events do not seem to be complete as they are missing the identifier or which sensor generated the event. 

What did you expect to happen?
Entity should be created for the lock/unlock and the identity of the sensor should be passed. 

driver logs attached.

### Device information

Manufacturer: Nexia Home Intelligence 
Model name:  VeriLock Translator / Product code 9125051
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version: Zwave-JS-UI from Homeassistant Community addon repository 1.15.8
`node-zwave-js` branch: 11.13.0
`zwave-js-ui` branch: 8.23.2

### Did you change anything?

yes (please describe)

### If yes, what did you change?

New install of Verilock translator.

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Never worked before but it apperars it was supported on smartthings with this groovy code.
https://github.com/erocm123/SmartThingsPublic/blob/master/devicetypes/erocm123/verilock-translator.src/verilock-translator.groovy

Also its supported on Hubitat and on Nexia's hub.

### Attach Driver Logfile

[zwavejs_2023-09-04.log](https://github.com/zwave-js/node-zwave-js/files/12518193/zwavejs_2023-09-04.log)

AlCalzone, 2023-09-05:
The logfile does not include a device interview, so I'm not sure what capabilities the device has. I'm assuming it doesn't support Door Lock CC, as that would require controlling a lock, and you're talking about sensors.
As such, there are no lock entities HA can expose.

Locking and unlocking is exposed via notification events, which are by definition stateless. Therefore, no values are created for those, unlike the window/door status which is a notification state. Instead, the events show up in HA's logbook. There's probably a way to make an entity from those via blueprints, but I don't know how. You'll have to ask in the HA forums or Discord.

Unfortunately, you're also running into [this bug](https://github.com/zwave-js/node-zwave-js/issues/5399). The notification events exposed by Z-Wave JS do not include the endpoint information and appear to come from the device, although the device sends them from an endpoint:

```
2023-09-05T02:32:05.317Z DRIVER « [Node 219] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      5
                                    │ destination: 0
                                    └─[NotificationCCReport]
                                        notification type:   Access Control
                                        notification status: 255
                                        notification event:  Manual lock operation
2023-09-05T02:32:05.318Z CNTRLR   [Node 219] [Notification]
                                    type:  Access Control
                                    event: Manual lock operation

```
There's already [a fix](https://github.com/zwave-js/node-zwave-js/pull/6083), but it is a breaking change, so I've put it on hold until I do another major release.

mt98987987, 2023-09-05:
Adding some more context for the sake of completeness here. I also added a new log which includes an interview.

>  I'm assuming it doesn't support Door Lock CC, as that would require controlling a lock, and you're talking about sensors.
This device cannot control the locks. It supports Verilock sensors made by Honeywell for Andersen. They have three models available. 

Verilock sensor types:
1. Window latches for double hung windows that replace the lock. These can show open/close state and also locked/unlocked state. The lock state is a sensor only, it cannot be controlled. 
2.  Window latches for awning and  windows that replace the lock. These can show open/close state and also locked/unlocked state. The lock state is a sensor only, it cannot be controlled.
3. A door lock for patio sliding doors. These can show open/close state and also locked/unlocked state. The lock state is a sensor only, it cannot be controlled.

[zwavejs_2023-09-05.log](https://github.com/zwave-js/node-zwave-js/files/12528567/zwavejs_2023-09-05.log)

AlCalzone, 2023-09-06:
I went ahead and finished a beta release of `v12`. When applications pick that up, your issue should be fixed.

AlCalzone, 2023-09-29:
v12 is now released with the fix for this

closed 2023-09-29 by AlCalzone