zwave-js/zwave-js #6248: Corrupt Hub Configuration

pjgolian, 2023-09-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I upgraded to version 0.1.88 to 0.1.89 today.  After upgrading none of my z-wave devices would respond.  I have two zwave hubs.  The first was using the default built in z-wave integration (using whatever the default name) and the 2nd using JS (using the name Main).  I tried restoring from a full backup made this morning.  That brought back my Main hub, but the secondary hub (the default zwave integration) is not working and is now called "Z-Wave JS".  

[image]

I am not sure how to fix this issue.  The main hub (which was using JS) is working correctly, but the secondary hub (default z-wave integration) is no longer working and was never setup to using JS.

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Upgraded to 0.1.89

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Was working on 0.1.88 before upgrading to 0.1.89.

### Attach Driver Logfile

I'm not sure where the log file is.

pjgolian, 2023-09-06:
I tried the steps listed in the bot response, but I can get into JS UI since it will not load.

pjgolian, 2023-09-06:
I was able to bring back the secondary hub by going into the Z-Wave add on and changing it from using JS back to the default zwave interface.  However, JS UI still fails to load for the Main hub.

AlCalzone, 2023-09-06:
Unfortunately I'm unable to provide application support (e.g. for Home Assistant) here. Please ask on the HA forums or Discord.

closed 2023-09-06 by AlCalzone