zwave-js/zwave-js #6259: 4 devices stuck on "node info" for a while (2 days)

rezzalex, 2023-09-07:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.24.2.1caa735

### ZwaveJS version

11.14.0

### Describe the bug

After 2 majors events, a huge number of my nodes where not correctly rediscoverd, 1 or 2 stucked on "Protocol Info" state.
Now, after physicakky unplugin / plugin the Controller, force to have most of the devices having a "real road" on the network graph, clear retain, force redisocover, ping, re-interview the nodes... I still have nodes that can not be recognised correctly and be used.

the 2 almost simultaneous events are :   
  - migrating to a recent bugged version (8.24 ?)
  -  having a LOT of power failure / power restart during 1 or 2 days, impacting the host and its zwave controller, and 2 power (non battery) nodes, one of them still having obvious discovery issues (node 6), the other seems OK but in fact is badly recognized (node 62),. The other problematic nodes are battery devices where I could have suspicions of lack of battery... but it seems their respective latest activities are not so old, so... coud be OK q battery level perspective

[image]

[image]

Could you please indicate me how to provide the necessary nodes ?

### To Reproduce

I don't know...

### Expected behavior

having all devices recognised correctly

### Additional context

my host is shutdowned every day at midnight and restart early in the morning

AlCalzone, 2023-09-07:
Make sure to include the startup of Z-Wave JS UI in the log (should normally be the case when you log to file).

AlCalzone, 2023-09-07:
No need. Activating the driver logs restarts already.

rezzalex, 2023-09-07:
@AlCalzone , here you are

[z-ui_2023-09-07.log](https://github.com/zwave-js/node-zwave-js/files/12550074/z-ui_2023-09-07.log)

candrea77, 2023-09-07:
Hi guys,
I've tried to upgrade from zwave-js-ui: 8.23.0.73606d4 zwave-js: 11.12.0 already two times (going to 8.23.2 and 8.24) but everytime I've to rollback to 8.23 (zwavejs 11.12) 'couse my network being stuck after some hours (can't turn on / off any devices from both zwavejs-ui and/or homeassistant)

 I've got same stick as @rezzalex , so maybe this is the same problem.

In case : here you are my log.
- Updated docker image @2023-09-04T18:39:41.473.
- Found network stuck in the morning when my ha routine try to turn on nodeid 048 @2023-09-05T04:05:00.004 (Light don't turn on even after multiple retry)
- Roll back to 11.12 and problems gone @2023-09-05T04:17:50.996 .

I'm on GMT+2 so logs are splitted at 02.00

[zwavejs_2023-09-04.log.gz](https://github.com/zwave-js/node-zwave-js/files/12551248/zwavejs_2023-09-04.log.gz)
[zwavejs_2023-09-05.log.gz](https://github.com/zwave-js/node-zwave-js/files/12551251/zwavejs_2023-09-05.log.gz)

Regards

rezzalex, 2023-09-08:
Hello.
Could you please have a look ?

Should I try to exclude and re-include my problematic nodes ?
Restore something?
Or do I have a some hardware failures due to power outages ?

rezzalex, 2023-09-08:
please help, it is getting worse and worse.  I don't want to reinstall it all : 

Here is the state after revert back to 8:23
: 
[image]

AlCalzone, 2023-09-08:
@rezzalex what you posted are not [driver logs](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs). Please read the link again.

AlCalzone, 2023-09-08:
@candrea77 There's definitely something weird going on. Looks like the primary queue is stuck. Could you execute the following code using Z-Wave JS UI's **driver function** feature in the advanced menu when it happens again?

```ts
driver.driverLog.sendQueue(driver.queue);
```

This should add a few lines to the logfile, showing what's currently in the queue.

AlCalzone, 2023-09-08:
Narrowed it down to some time between 21:07 and 21:59, maybe even 21:07 - 21:09. This is the last time a sleeping node gets actively sent to sleep again:
```
2023-09-04T21:07:17.560Z DRIVER « [Node 139] [REQ] [ApplicationCommand]
                                  └─[WakeUpCCWakeUpNotification]
2023-09-04T21:07:17.560Z CNTRLR « [Node 139] received wakeup notification
2023-09-04T21:07:17.561Z CNTRLR   [Node 139] The node is now awake.
...
2023-09-04T21:07:19.354Z CNTRLR » [Node 139] Sending node back to sleep...
2023-09-04T21:07:19.355Z DRIVER   all queues busy
2023-09-04T21:07:19.355Z SERIAL » 0x010900138b0284082400c4                                            (11 bytes)
2023-09-04T21:07:19.355Z DRIVER » [Node 139] [REQ] [SendData]
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[WakeUpCCNoMoreInformation]
```

After that, this does not happen anymore:

```
2023-09-04T21:09:51.128Z DRIVER « [Node 098] [REQ] [ApplicationCommand]
                                  └─[WakeUpCCWakeUpNotification]
2023-09-04T21:09:51.128Z CNTRLR « [Node 098] received wakeup notification
2023-09-04T21:09:53.391Z SERIAL « 0x0121000400a8199f03420093db5a2fc142a732a095eb57dda0527dd4baf97c147 (35 bytes)
                                  e00c4
2023-09-04T21:09:53.392Z SERIAL » [ACK]                                                                   (0x06)
```

Not sure yet why, will have to investigate. Seeing the send queue log in combination with the full logs would definitely help narrow this down.

candrea77, 2023-09-08:
Ok

I suppose that doing this command now (I'm on 11.12 with network status good) don't be helpfull
Anyway .... this is the logs obtained (command executed 3 times)  : 

```
2023-09-08T22:01:34.014Z DRIVER   Send queue:                                                       (2 messages)
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]

2023-09-08T22:04:38.022Z DRIVER   Send queue:                                                       (2 messages)
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]

2023-09-08T22:06:12.248Z DRIVER   Send queue:                                                       (2 messages)
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]
```

If I've understood correctly tomorrow i will :

- update zwavejs to 11.13.x ( or something newer)
- wait until network hang
- execution the code to get the command queue
- update this 3d with new logs

So, I will update this 3d as soon as I get this new data.

Thanks for the support !

AlCalzone, 2023-09-09:
```
2023-09-08T22:04:38.022Z DRIVER   Send queue:                                                       (2 messages)
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]
                                  · [REQ] SendData (Security2CCMessageEncapsulation) [Node 168, Asleep]
```
that looks normal though. You're sending a command to a sleeping node, so that gets queued until it wakes up.
Or did you try to send a command to a different device?

rezzalex, 2023-09-12:
> @rezzalex what you posted are not [driver logs](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs). Please read the link again.

[zwavejs_2023-09-12.log](https://github.com/zwave-js/node-zwave-js/files/12583078/zwavejs_2023-09-12.log)

AlCalzone, 2023-09-12:
@rezzalex there's **a lot** of corrupted data coming from your Z-Wave stick.
For example:
```
SERIAL « [DISCARDED] invalid data 0x0a9880d1687c8505270f36da                 (12 bytes)
```
In normal operation this shouldn't be happening at all, your log has it over 170 times.

Have you unplugged and re-plugged the stick yet? If not, that could be worth a try. Leave it unplugged for a minute or so. Maybe use a different USB port?

rezzalex, 2023-09-12:
HI @AlCalzone , I did previously unplugged the stick with no results, .... but the time unplugged was not enough. Unplugging it for 3 to 4 minutes did the trick ! thanks a lot

candrea77, 2023-09-13:
Hi guys,
I also did a lot of testing for the previous zwave network glitch.

I've upgraded to latest version (and also to intermediate ones) and never had the problem again.

By my point of view @rezzalex you can safely close this issue.
Thank you all

closed 2023-09-13 by rezzalex