zwave-js/zwave-js #6265: Network randomly goes unstable with no commands being received or sent

rohankapoorcom, 2023-09-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I'm not able to track down specifically what is causing the issue. But consistently at arbitrary times (sometimes it takes 1 day, sometimes it takes 3, but definitely happens at least 1x a week) in the last month, I've noticed that the entire network stops receiving or sending any commands.

What do you observe?
When this happens, I see the Dropped RX Messages in Z-Wave JS UI climb up constantly (as sensors are trying to provide updates). Looking in the log file every incoming message has the following error:

```
2023-09-10T18:27:30.500Z SERIAL « 0x010e0004003705310505012fbf000067                                  (16 bytes)
2023-09-10T18:27:30.500Z DRIVER   Dropping message because it could not be deserialized: The command class unkno
                                  wn (0x05) is not implemented (ZW0303)
2023-09-10T18:27:30.501Z SERIAL » [ACK]                                                                   (0x06)
```

Once this occurs, commands from Z-Wave JS UI (or Home Assistant) to the network do not get received as well.

Going into Z-Wave JS UI settings and clicking save (which reloads Z-Wave JS) fixes the problem immediately.

What did you expect to happen?
Messages should continue to be processed and connectivity should be fine.

Steps to reproduce the behavior:
I do not have a reproducible way of doing so, it just happens on it's own, psuedo randomly.

### Device information

N/A

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.23.1.77058e8
zwave-js: 11.13.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I've run with the same Z-Wave JS setup for several years now. I did add some new devices from Inovelli (VZW31-SN) a couple of months ago (July). I had previously been running a beta version of this device with no issues since December.

### Attach Driver Logfile

[zwavejs_2023-09-10.log](https://github.com/zwave-js/node-zwave-js/files/12569925/zwavejs_2023-09-10.log)

AlCalzone, 2023-09-11:
Looks like the controller starts hanging in the middle of a transmission, then restarts itself. However, Z-Wave JS still thinks the controller is communicating with 16-bit node IDs, although it reverted to 8-bit IDs after the restart.

```
2023-09-10T11:27:13.003Z DRIVER « [REQ] [SerialAPIStarted]
                                    wake up reason:        SoftwareReset
                                    watchdog enabled:      false
                                    generic device class:  0x02
                                    specific device class: 0x01
                                    always listening:      false
                                    supports Long Range:   false
2023-09-10T11:27:13.003Z DRIVER   handling request SerialAPIStarted (10)
2023-09-10T11:27:13.004Z DRIVER     no handlers registered!
2023-09-10T11:27:28.306Z SERIAL « 0x01130004004d0a32022174000000000000ad000067                        (21 bytes)
2023-09-10T11:27:28.306Z DRIVER   Dropping message because it could not be deserialized: The command class unkno
                                  wn (0x02) is not implemented (ZW0303)
2023-09-10T11:27:28.306Z SERIAL » [ACK]                                                                   (0x06)
2023-09-10T11:27:49.052Z SERIAL « 0x011b0004006c129f030100367791c58df8df8aff9dfca91636a3000082        (29 bytes)
2023-09-10T11:27:49.053Z DRIVER   Dropping message because it could not be deserialized: The command class unkno
                                  wn (0x03) is not implemented (ZW0303)
2023-09-10T11:27:49.053Z SERIAL » [ACK]                                                                   (0x06)
2023-09-10T11:27:49.157Z SERIAL « 0x01180004006c0f9f030000629ad5a2f5ae6f36c468c5a8000050              (26 bytes)
2023-09-10T11:27:49.158Z DRIVER   Dropping message because it could not be deserialized: The command class unkno
                                  wn (0x03) is not implemented (ZW0303)
2023-09-10T11:27:49.158Z SERIAL » [ACK]                                                                   (0x06)
2023-09-10T11:27:58.306Z SERIAL « 0x01130004004d0a32022174000000000000ad000067                        (21 bytes)
2023-09-10T11:27:58.306Z DRIVER   Dropping message because it could not be deserialized: The command class unkno
                                  wn (0x02) is not implemented (ZW0303)
```

closed 2023-09-13 by zwave-js-bot

rohankapoorcom, 2023-09-21:
Nice, Just updated to it. I'll report back if the problem reoccurs

Thanks!

PR #6271 (2023-09-13): fix: switch to 16 bit node IDs again when Serial API restarts unexpectedly
fixes: #6265