zwave-js/zwave-js #6268: Fakro AMZ Z-wave losing state after 8.18.1

bluediamond1984, 2023-09-11:


bluediamond1984, 2023-09-11:
@AlCalzone, here is the log.
[zwavejs_2023-09-09.log](https://github.com/zwave-js/node-zwave-js/files/12579099/zwavejs_2023-09-09.log)

AlCalzone, 2023-09-12:
Do you know if it ever reports the final status on its own? Your log doesn't look that way, although the manual indicates that it should:
[image]

Can you share a screenshot of the **Groups** tab?

bluediamond1984, 2023-09-12:
It doesn't. When press open/close there is a brief moment of the new state and then it returns `null`. A push of the refresh button is always needed.

My groups from the device are empty. And the is nothing I can add.
[image]

That got me thinking and I have now added polling for Current value with an interval of 10s. Now it's reporting the state after 20 seconds. So that part is working.
[image]

But the state is still returning `null` directly after open/close action.

AlCalzone, 2023-09-12:
Polling isn't ideal. Looks like the device information wasn't requested fully. Can you re-interview the device and share another driver log of that?

bluediamond1984, 2023-09-13:
Devices are re-interviewd and I have also upgraded to 11.14.2. See attachment for debug log. I have removed polling from the settings in the second try (I had forgotten to save the settings).
[zwavejs_2023-09-13.log](https://github.com/zwave-js/node-zwave-js/files/12600208/zwavejs_2023-09-13.log)

In groups nothing has changed.
Polling on it's own doesn't work.
The Current value has changed and isn't returning `null` anymore when I open/close with Binary Switch or Multilevel Switch. A manual refresh is still needed.
When I change state from HA, then Current value is still returning `null` after 5 seconds.

AlCalzone, 2023-09-20:
Ok, looks like the device in question is the old (non Z-Wave+) variant here: https://products.z-wavealliance.org/products/677?selectedFrequencyId=-1

That doesn't support associations, so you must probably poll it for changes made on the device itself. I'll look into handling the "unknown/null" state after controlling it.

AlCalzone, 2023-09-20:
Info for future me: 
- Motor Control Class B, only endpoint aware, not position aware

bluediamond1984, 2023-09-21:
Thanks @AlCalzone for your support. Polling works and it is what it is. I look forward for a solution you can come up for the null state.