zwave-js/zwave-js #6273: Inovelli VZW31-SN Firmware 1.2 New Parameters

DAVe3283, 2023-09-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

#### What causes the bug?

Firmware 1.2 (https://community.inovelli.com/t/red-series-2-1-firmware-changelog-vzw31-sn/14579) was recently promoted to stable (zwave-js/firmware-updates#83). It adds 2 new parameters (temperature and overheat) and an option from 1.1 (leading or trailing edge dimming). These options don't have config values in this repo.

#### What do you observe?

The temperature parameter shows up, but it is not being reported as a measurement, and the temperature units are missing. This causes Home Assistant to display it is a string, instead of a number and recording the history in a graph.
[image]

Switching between leading and trailing edge is also somewhat ambiguous.
[image]
On other parameters, the values would be labeled in a drop-down, e.g. `[0] Leading Edge`, `[1] Trailing Edge`.

#### What did you expect to happen?

The new parameters should show up with the correct units as the other parameters do. This would let the temperature show up as a history graph like other Z-Wave devices reporting a temperature have.

#### Steps to reproduce the behavior:

1. Update a VZW-31SN to firmware 1.2
2. Enable one of the new parameters (e.g. Internal Temperature) in Home Assistant
3. Note how the parameter has no units and doesn't get a history graph.

### Device information

Manufacturer: Inovelli
Model name: Red Series 2-1 Dimmer (VZW31-SN)
Node ID in your network: 9

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Z-Wave JS UI: 8.24.2
Z-Wave JS: 11.14.0
Home Assistant Add-On Version: 1.15.9

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Do you need a log of a re-interview to see the new parameters? Let me know what you need and I will provide it.

InovelliUSA, 2023-09-11:
For P26 can we NOT have it as a selectable option in the GUI or have it read only? Changing this setting remotely can damage the device if an improper load is connected. It is ok to change it via combo button or to send it as a custom configuration parameter from the control panel:

[image]

Just want to be cautious that people don't willy nilly change it. The combo button is:

Hold down + press config 13x = trailing edge
Hold up + press config 13x = leading edge

AlCalzone, 2023-09-12:
Absolutely!

AlCalzone, 2023-10-12:
Parameter 26 added in #6391

InovelliUSA, 2023-10-13:
@AlCalzone Thanks! I see the note you placed in the file.