zwave-js/zwave-js #6281: Wrong Home Health, VOC level status values

cathrinevaage, 2023-09-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
I believe the notification parameter value is not correctly deserialized. 

**What do you observe?**
When listening for changes on my nodes that emits `Home Health`, `VOC level status` reports, the value is wrong.
The values I've observed are 1538 and 1539.

**What did you expect to happen?**
I expected to get any of the values specified in the [notifications.json:L1123](https://github.com/zwave-js/node-zwave-js/blob/b45a75f10f3b21e15e58b619bdfc092cac0105bd/packages/config/config/notifications.json#L1123) config. 0x01, 0x02, 0x03, or 0x04.

### Device information

Manufacturer: Eurotronic
Model name: Air Quality Sensor Z-Wave Plus
(https://github.com/zwave-js/node-zwave-js/blob/5df7cd4f1156b17659c9d725b8d29b0679402a4f/packages/config/config/devices/0x0148/eur_airquality.json)
Node ID in your network: 16 and 17

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [X] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 11.14.2

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Don't know, first time I'm trying it since [support for notification event parameters were added in February](https://github.com/zwave-js/node-zwave-js/issues/264).

### Attach Driver Logfile

[test.log_2023-09-13.log](https://github.com/zwave-js/node-zwave-js/files/12606777/test.log_2023-09-13.log)

cathrinevaage, 2023-09-14:
Here's the same log file again, with hopefully the desired file name this time.
[zwavejs_2023-09-13.log](https://github.com/zwave-js/node-zwave-js/files/12606828/zwavejs_2023-09-13.log)

AlCalzone, 2023-09-15:
Looks like `getNotificationEnumBehavior` is only used in defining the possible states, but not when actually setting the value in https://github.com/zwave-js/node-zwave-js/blob/5df7cd4f1156b17659c9d725b8d29b0679402a4f/packages/cc/src/cc/NotificationCC.ts#L410

PR #6282 (2023-09-15): fix: handle notification enum behavior "replace" correctly
We already had a test for this, but the test didn't match the expectations explained in the comment.
fixes: #6281
closed 2023-09-15 by zwave-js-bot