zwave-js/zwave-js #6284: Super slow network updating the same value many times

Jordenjj, 2023-09-18:
Hello,
My network has become really slow, and I've noticed for many devices that the exact same value is updated a bunch of times. Slowing down the network. I do not have any idea what could cause this. When I turn on the lights, node 008 for instance shows
```
[Node 008] Value updated: 37-1-currentValue true => true
```
a total of 24 times, while it even is already the same.

```
true => true
```
shows up a total of 159 times.
These lights grouped together a staggering 28 seconds to all power on.
Any idea what this might be?

Full logs of switching ON all devices once:
```

Debug
Scan completed
Filter logs
Type to filter logs, case sensitive
2023-09-17 14:36:06.871 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:06.907 INFO Z-WAVE: [Node 006] Value updated: 37-1-targetValue false => true
2023-09-17 14:36:06.908 INFO Z-WAVE: [Node 006] Value updated: 37-1-currentValue false => true
2023-09-17 14:36:06.938 INFO Z-WAVE: [Node 006] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:07.029 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:07.346 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:07.931 INFO Z-WAVE: [Node 003] Value updated: 37-1-targetValue false => true
2023-09-17 14:36:07.932 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue false => true
2023-09-17 14:36:08.003 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:08.045 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:08.105 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:08.176 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:08.176 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 0 => 10.05
2023-09-17 14:36:08.258 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:08.259 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:08.418 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:08.557 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:08.557 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:08.603 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:08.603 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:08.638 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:08.870 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:08.871 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:08.917 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:08.959 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:08.960 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:09.005 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:09.007 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:09.150 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.209 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.234 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.327 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.436 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.465 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.562 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.622 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.648 INFO Z-WAVE: [Node 003] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.692 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.844 INFO Z-WAVE: [Node 006] Metadata updated: 50-1-value-66049
2023-09-17 14:36:09.844 INFO Z-WAVE: [Node 006] Value updated: 50-1-value-66049 0.8 => 34.6
2023-09-17 14:36:09.878 INFO Z-WAVE: [Node 002] Value updated: 37-1-currentValue true => true
2023-09-17 14:36:09.932 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:09.933 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:09.970 INFO Z-WAVE: [Node 025] Metadata updated: 50-0-value-66049
2023-09-17 14:36:09.970 INFO Z-WAVE: [Node 025] Value updated: 50-0-value-66049 0 => 26.33
2023-09-17 14:36:09.993 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:09.994 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
2023-09-17 14:36:10.043 INFO Z-WAVE: [Node 019] Metadata updated: 50-0-value-66049
2023-09-17 14:36:10.044 INFO Z-WAVE: [Node 019] Value updated: 50-0-value-66049 10.05 => 10.05
[... 427 lines cut ...]
2023-09-17 14:36:32.434 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.462 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.463 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.526 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.527 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.684 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.685 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.712 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.713 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.850 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.850 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.866 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.867 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:32.915 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:32.917 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:33.065 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:33.065 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:33.225 INFO Z-WAVE: [Node 007] Metadata updated: 50-1-value-66049
2023-09-17 14:36:33.226 INFO Z-WAVE: [Node 007] Value updated: 50-1-value-66049 24.1 => 24.1
2023-09-17 14:36:33.584 INFO Z-WAVE: [Node 008] Metadata updated: 50-1-value-66049
2023-09-17 14:36:33.584 INFO Z-WAVE: [Node 008] Value updated: 50-1-value-66049 62 => 62.1
2023-09-17 14:36:34.331 INFO Z-WAVE: [Node 005] Metadata updated: 50-1-value-66049
2023-09-17 14:36:34.332 INFO Z-WAVE: [Node 005] Value updated: 50-1-value-66049 0.8 => 6.7
2023-09-17 14:36:34.681 INFO APP: GET /health/zwave 301 0.321 ms - 191
```

Jordenjj, 2023-09-18:
Strange. Here it is. 
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12646803/zwave-js-ui-store.zip)

AlCalzone, 2023-09-20:
This looks like a connectivity issue where the command is received by the controller, but devices don't receive the acknowledgement for that, causing them (or repeaters along the route) to re-transmit.

Node 10 (and at one point Node 22) only have a route that uses 9.6kbit/s, so there's definitely a problem. 

I'll start with a few links for you to work through.
1. Put the stick on an extension cable if you haven't already: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
2. Figure out which nodes have a good connection and which don't: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength

Looks like you already have a priority route to at least one device, so you probably have played with this a bit already?

Other than that: 700 series USB sticks are often reported to be problematic for EU users (terrible range). On top of that, some users have also reported that the 7.19.x firmware has terrible range for them. You have both.

Do you have a 500 series stick you could go back to, e.g. Aeotec Z-Stick 5(+) with firmware 1.2?

Jordenjj, 2023-09-20:
> Put the stick on an extension cable if you haven't already

Its on a extension cable currently

>Looks like you already have a priority route to at least one device, so you probably have played with this a bit already?.

Yes. Unfortunately to no avail.

>Other than that: 700 series USB sticks are often reported to be problematic for EU users (terrible range)

Shucks... here I was thinking newer = better

>Do you have a 500 series stick you could go back to, e.g. Aeotec Z-Stick 5(+) with firmware 1.2?

I don't, but I can purchase one. Unsure whether a NVM backup works when downgrading 700 > 500

Thank you.

AlCalzone, 2023-09-20:
Zwave-js converts the NVM so you can go to 500 and back. The firmware must be new enough though, so if the Z-Stick 5 isn't using 1.2, you need to upgrade it first.
Also make sure to get the 5+, not the 5. Some shops don't list this clearly.

Jordenjj, 2023-09-20:
[image]
This is the network at the moment

AlCalzone, 2023-09-21:
Walk/Downstairs seems like a problem. It only reaches 40kbit/s and several nodes are routing through it with even worse connectivity (9.6k).
I'd also look at the Living Room Door and Vinyl Player (also 9.6k), the rest may be okay.

Jordenjj, 2023-09-21:
> I'd also look at the Living Room Door and Vinyl Player (also 9.6k), the rest may be okay.

When you say that do you mean look into creating manual routes?

AlCalzone, 2023-09-21:
Maybe, but not necessarily. Looks like *Walk/Downstairs* already has a direct connection, but a suboptimal one. Maybe there's a good repeater available that's not being used. Maybe its location is just not good. Maybe it needs another repeater inbetween. Or maybe it is actually not a big issue but the connection from it to the nodes "behind" it is.

closed 2023-10-09 by zwave-js-assistant[bot]