zwave-js/zwave-js #6285: Missing device configuration: Mitsubishi Z-Wave Adaptor FHAZWAC0603P

jaredAmpo, 2023-09-18:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Mitsubishi Z-Wave Adaptor FHAZWAC0603P

### Manufacturer ID

316

### Product Type

1

### Product ID

89

### Firmware Version

1.4.1

### Do you have a manual?

[FHAZWAC0603P manual-V2-20210125.pdf](https://github.com/zwave-js/node-zwave-js/files/12645843/FHAZWAC0603P.manual-V2-20210125.pdf)

Device is able to be detected as `climate` type, however I am unable to adjust the temperature. Below is the snippet of logs,
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 227, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1969, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2006, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 617, in async_service_temperature_set
    await entity.async_set_temperature(**kwargs)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/climate.py", line 465, in async_set_temperature
    setpoint: ZwaveValue = self._setpoint_value_or_raise(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/climate.py", line 213, in _setpoint_value_or_raise
    raise ValueError("Value requested is not available")
ValueError: Value requested is not available
```

Would appreciate any insights to this issue, thanks!

AlCalzone, 2023-09-20:
This is an issue in Home Assistant, and seems to be closely related to https://github.com/zwave-js/certification-backlog/issues/2

AlCalzone, 2023-09-20:
/cc @marcelveldt

closed 2023-09-24 by zwave-js-assistant[bot]