zwave-js/zwave-js #6290: Support BinarySwitchCCSet, ThermostatModeCCSet

XLDom, 2023-09-18:
**Is your feature request related to a problem? Please describe.**
I've got a HRT4-ZW / SRT321 and the value of switch and mode is not present
This is this device : https://devices.zwave-js.io/?jumpTo=0x0059:0x0001:0x0003:0.0

**Describe the solution you'd like**
Add support for BinarySwitchCCSet, ThermostatModeCCSet

**Additional context**
Here is log file with the message : TODO: no handler for application command for BinarySwitchCCSet, ThermostatModeCCSet
[zwavejs_2023-09-18.log](https://github.com/zwave-js/node-zwave-js/files/12653305/zwavejs_2023-09-18.log)

Exemple : 
```
2023-09-18T20:09:45.235Z SERIAL « 0x01090004080203250100df                                            (11 bytes)
2023-09-18T20:09:45.238Z SERIAL » [ACK]                                                                   (0x06)
2023-09-18T20:09:45.241Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  │ type: multicast
                                  └─[BinarySwitchCCSet]
                                      target value: false
2023-09-18T20:09:45.242Z CNTRLR « [Node 002] TODO: no handler for application command
```
and
```
2023-09-18T20:09:45.147Z SERIAL « 0x01090004000203400100b2                                            (11 bytes)
2023-09-18T20:09:45.151Z SERIAL » [ACK]                                                                   (0x06)
2023-09-18T20:09:45.154Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  └─[ThermostatModeCCSet]
                                      mode: Off
2023-09-18T20:09:45.156Z CNTRLR « [Node 002] TODO: no handler for application command
```

Thanks for all

AlCalzone, 2023-09-19:
The commands in your log are meant to control devices, but they are received by zwave-js.
I'm inclined to say that's not what you meant. Will look at the log tomorrow when I'm back at work.

mrgreenfr, 2023-09-19:
Hello thank you. I understand what you said. Before I was using openzwave and I was able to retrieve the values of these commands (to know if the thermostat was requesting heating or not for example). Tell me if you need more information.

XLDom, 2023-09-19:
Hello thank you. I understand what you said. Before I was using openzwave and I was able to retrieve the values of these commands (to know if the thermostat was requesting heating or not for example). Tell me if you need more information.

AlCalzone, 2023-09-20:
I see, this is https://github.com/zwave-js/node-zwave-js/issues/5937 then. I'll add your use case to that issue.

closed 2023-09-20 by AlCalzone