zwave-js/zwave-js #6298: Razberry Pro 7 and Vision ZD2102 sensor

vaxjool, 2023-09-21:
Hi, 

My Vision ZD2102-EU [(documentation),](https://manual.zwave.eu/backend/make.php?lang=en&sku=B_VIS_ZD2102&cert=) Razberry 7 Pro, doens't work with zwave-js-ui/zwavejs2mqtt after version 8.21.

After migration/re-creating the container using the :latest -version, all my battery operated door sensors stop working, with any version of zwavejsui, greater than version 8.21. It doesn't register open/closed state. I can ping it if I keep it "alive". If re-interviewing the configuration is corrupt/non existing. 

Version 8.21 works perfectly with the exact same hardware and devices.

Any suggestions as to why this is the case?

br

vaxjool, 2023-10-02:
Hi, sorry for the delay, it's just a crazy time atm. Thanks for looking into this, appreciate it!

I've attached both files and I re-interviewed device 23. 

As you can see, after the re-interview the device id is wrong, using the old 8.21.2.c version it is: 265-258-8193 (0x0109-0x2001-0x0102) and it doesn't register state changes in newer versions.

[zwavejs_2023-10-02.log](https://github.com/zwave-js/node-zwave-js/files/12785130/zwavejs_2023-10-02.log)
[z-ui_2023-10-02.log](https://github.com/zwave-js/node-zwave-js/files/12785132/z-ui_2023-10-02.log)

AlCalzone, 2023-10-06:
This looks like a bug in the controller firmware. Z-Wave JS enables 16-bit node IDs, which the controller claims to support, but once enabled it sets the node ID of all incoming commands to 1. Therefore, the responses from the node are not detected.

I don't specifically see this issue mentioned on the changelog, but there are a few firmware upgrades available: https://z-wave.me/support/uzbrazberry-firmwares/
Currently I'd recommend not going above **7.28 / 07.17.02.01 SDK 7.17.02** though. The newer Z-Wave SDKs have a few glaring bugs.

vaxjool, 2023-10-08:
Ok, thanks, I'll report back as soon as I've tested with newer firmware. I'll try to test this during the week.

closed 2023-10-12 by zwave-js-assistant[bot]