zwave-js/zwave-js #6311: Extraction script from OpenSmartHouse needs adjustments

apella12, 2023-09-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Running this script; yarn run config import -s oh -Dd --ids xxxx

What do you observe?
Script wrongly sets allowManualEntry=true when the OpenSmartHouse "limit to options" is true.  In the OpenSmartHouse context the limit to options =true means that only the options shown are allowed. In the zwave-js context this means allowManualEntry=false.  Also the PR checker informed me allowManualEntry=true is not allowed (I'm assuming it is deemed unnecessary), so script should ideally not add the allowManualEntry parameter if the OpenSmartHouse limit to options is false.

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 12.0.0-beta.3 

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

This is script issue.  The only report is DONE;  DONE

AlCalzone, 2023-09-23:
Happy to take a PR if you feel comfortable with that. If not I can do it next week.

apella12, 2023-09-23:
I'll haven't located the file yet, but can try and take a look for a PR.  Just wanted to note the issue first.

PR #6313 (2023-09-25): fix: meaning of `limit_options` in OpenSmartHouse import script
fixes: #6311
closed 2023-09-25 by zwave-js-bot