zwave-js/zwave-js #6316: Error: Controller is unable to transmit

calanthe23, 2023-09-25:
```
zwave-js-ui: 8.26.0
zwave-js: 11.14.3
```
```
zwave stick: ZST39 LR
FW: v1.20
SDK: v7.19.3
```
```
Home Assistant 2023.9.2
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest 
```

I am running the latest version of zwave JS UI.  Yesterday, I noticed a red box in the upper-right corner of the zwave JS UI control panel noting **"Controller is unable to transmit"**.  Most of the nodes in my 13 node mesh are still functioning, and I can receive data from them.  When initially troubleshooting the issue, I attempted to re-interview two of my nodes, and they have been stuck on "NodeInfo" for a considerable amount of time.  If I attempt to execute a new Inclusion, it ends immediately.

What logs or information may I provide to further troubleshoot this issue?  Thanks!

AlCalzone, 2023-09-25:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

AlCalzone, 2023-09-26:
Firmware v7.19.x (and maybe others) have a bug that causes this behavior. The upcoming v12 release of `node-zwave-js` will work around it by automatically soft-resetting the controller.
Unfortunately to fully fix this, the Z-Wave firmware needs to be fixed.

AvienseMalta01, 2023-09-26:
I have the same with Aeotec Z-Stick 7, FW Version 7.19.2.  The entire network stops - the controller still receives a lot of messages (and I mean a lot - devices seem to keep sending probably because they receive no acknowledgement) but I seem to be unable to control anything, and nothing seems to be being transmitted to the individual devices (I have a lot of devices though - 91 top be precise).  Most sensors seem to be getting a smiley face icon on the Compact View, seeming to be stuck in "awake" mode.

> Firmware v7.19.x (and maybe others) have a bug that causes this behavior. The upcoming v12 release of `node-zwave-js` will work around it by automatically soft-resetting the controller. Unfortunately to fully fix this, the Z-Wave firmware needs to be fixed.

Too bad though - does that mean we have to wait for Aeotec to fix their firmware?  Do they even know about this at all?  Who knows when such a fix will be forthcoming...  Also - someone above mentioned Zooz with same problem though.

Also - wouldn't a soft reset ruin any automations running in HASS at the time?

calanthe23, 2023-09-26:
> Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

I turned on debug logs & log to file.  I will capture logs until I note that the controller returns to a state of 'unable to transmit' and I will provide them at that time.

I also reached out to Zooz support in regards to the problem.

AlCalzone, 2023-09-26:
> Also - wouldn't a soft reset ruin any automations running in HASS at the time?

No, Z-Wave JS detects the necessity for this automatically, then resumes the previous command afterwards.

We have an [open issue](https://community.silabs.com/s/question/0D58Y0000ArDLnLSQW/sdk-719x-controller-locks-up-during-transmission-not-acknowledging-incoming-frames-missing-senddata-callback) with Silabs who provide the base firmware - Zooz just make minor modifications to that. Unfortunately we'll have to wait for Silabs to figure this out.

> But the network was completely ok when I was still on 1.15.7

Any chance you had a lot of randomly dead nodes during that time?

AlCalzone, 2023-09-26:
You were most likely seeing the same underlying issue (controller responding with a status of `Fail`), except that Z-Wave JS would mark the device as dead and move on. Now it considers this status as a problem of the controller not the device - specifically the controller not being able to transmit due to too much RF noise.

The problem is/was that Z-Wave JS assumes this situation would resolve itself on its own and retries indefinitely, but the firmware bug causes the same `Fail` status after a hanging transmission until the controller is soft-reset.
Anyways, version 12 that was released earlier today should fix this. AFAIK HA will pick it up in tomorrow's beta.

calanthe23, 2023-09-26:
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12731646/zwave-js-ui-store.zip)

calanthe23, 2023-09-26:
@AlCalzone Things ceased up again from the controller, and I have attached the logs.

madbrain76, 2023-09-26:
I am running into this too, on a ZST39 stick, FW: v1.20 SDK: v7.19.3.
It happens when I try to flash my most distant ZEN76 800LR switch. The node always dies during in the middle of the flashing process, and it results in this exact problem.

AlCalzone, 2023-09-27:
@calanthe23 same issue. Workaround is included in the next update.

Also, I'm told that Silabs are aware and have this issue as a top priority.

closed 2023-09-27 by AlCalzone

AlCalzone, 2023-09-27:
Please try again with the next version. The workaround works slightly different from a manual soft-reset.
If this continues to be an issue after updating, please open a new issue.

AvienseMalta01, 2023-09-27:
Thanks – will try that as soon as the new version becomes available.

bjornarfjelldal, 2023-11-07:
Hi! I am still having this problem on 9.3.1 / 12.2.3 on UZB-7 on 7.19.4, Latest HASS on Proxmox 8. The entire Z-Wave network becomes unresponsive until a soft reset. Allthough the log states traffic and status updates.

Anything I can do besides wait for SL to provide a firmware? I see the 7.20.x cannot even be flashed properly, and that it doesn't contain any major changes, so I guess this is not the one to update to?

For the record, I have read this; 

https://github.com/zwave-js/node-zwave-js/issues/6341

And set a daily restart of the Z-waveJS UI addon from HA. However this occurs more frequently than every 24 hours.

AlCalzone, 2023-11-07:
@bjornarfjelldal what you're experiencing might be fixed by https://github.com/zwave-js/node-zwave-js/pull/6473 which is included in the 12.3.0 release - except for the firmware bug itself. I've heard that Silabs are close to a release, but don't know any specifics.

bjornarfjelldal, 2023-11-07:
> @bjornarfjelldal what you're experiencing might be fixed by #6473 which is included in the 12.3.0 release - except for the firmware bug itself. I've heard that Silabs are close to a release, but don't know any specifics.

Thanks for the reply and for keeping my hopes up! :)

bjornarfjelldal, 2023-11-21:
Still getting this in 12.3.0 and the only fix is a restart of the addon - is this expected still because of the firmware bug?

2023-11-21 08:04:28.706 ERROR Z-WAVE-SERVER: Failed to send the command after 5 attempts (ZW0202)
```
ZWaveError: Failed to send the command after 5 attempts (ZW0202)
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:5240:23)
    at Driver.sendCommandInternal (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:5445:28)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:5556:15)
    at Proxy.set (/opt/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:264:24)
    at Proxy.<anonymous> (/opt/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:341:17)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:1150:39)
    at NodeMessageHandler.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:43)
    at Object.node (/opt/node_modules/@zwave-js/server/dist/lib/server.js:42:92)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:125:99)
    at WebSocket.<anonymous> (/opt/node_modules/@zwave-js/server/dist/lib/server.js:57:45)
2023-11-21 08:04:40.147 INFO Z-WAVE: [Node 109] Value notification: 91-0-scene-001 0
2023-11-21 08:04:40.170 ERROR Z-WAVE-SERVER: The node did not acknowledge the command (ZW0204)
ZWaveError: The node did not acknowledge the command (ZW0204)
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:5240:23)
    at Driver.sendCommandInternal (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:5445:28)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:5556:15)
    at Proxy.set (/opt/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:264:24)
    at Proxy.<anonymous> (/opt/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:341:17)
    at VirtualNode.setValue (/opt/node_modules/zwave-js/src/lib/node/VirtualNode.ts:172:39)
    at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/multicast_group/message_handler.js:13:50)
    at Object.multicast_group (/opt/node_modules/@zwave-js/server/dist/lib/server.js:43:108)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:125:99)
    at WebSocket.<anonymous> (/opt/node_modules/@zwave-js/server/dist/lib/server.js:57:45)
```

AlCalzone, 2023-11-21:
> Failed to send the command after 5 attempts

will require more complete driver logs to diagnose. Ideally open a new [support request](https://github.com/zwave-js/node-zwave-js/discussions/new?category=request-support-investigate-issue) and provide the information that is asked for.

> The node did not acknowledge the command

is a connectivity issue between the controller and the target node.