zwave-js/zwave-js #6323: Missing device configuration: Alloy Contact Sensor 2

swampjedi, 2023-09-26:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Alloy Contact Sensor 2

### Manufacturer ID

0x131

### Product Type

0x04

### Product ID

0x1118

### Firmware Version

2.19.0

### Do you have a manual?

[File.PDF](https://github.com/zwave-js/node-zwave-js/files/12720779/File.PDF)

apella12, 2023-09-26:
Same question here re: IDs

swampjedi, 2023-09-26:
> Same question here re: IDs

Fixed via edit, apologies for being imprecise.

apella12, 2023-10-11:
this should be closed with #6331 merge

closed 2023-10-11 by AlCalzone