zwave-js/zwave-js #6324: Missing device configuration: Alloy Motion Sensor (5 in 1)

swampjedi, 2023-09-26:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Alloy Motion Sensor (5 in 1)

### Manufacturer ID

0x131

### Product Type

0x04

### Product ID

0x1123

### Firmware Version

2.6.0

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

[File0001.PDF](https://github.com/zwave-js/node-zwave-js/files/12720801/File0001.PDF)

apella12, 2023-09-26:
This looks like a near clone of 
"manufacturer": "Shenzhen Neo Electronics Co., Ltd.",
"manufacturerId": "0x0258",
"label": "NAS-PD07ZU1",
"description": "5 in 1 PIR Motion Sensor",

but I'm confused by your MFG ID, and product ID.  The need to be 0x????.  Is mfg 0x0131 and product ID 0x0400 or 0x0004 or what?

swampjedi, 2023-09-26:
> but I'm confused by your MFG ID, and product ID. The need to be 0x????. Is mfg 0x0131 and product ID 0x0400 or 0x0004 or what?

Fixed via edit, apologies for being imprecise.

apella12, 2023-10-11:
this should be closed with #6331 merge

closed 2023-10-11 by AlCalzone