zwave-js/zwave-js #6328: Philip PSR03-C Panic Button - No MQTT "button" entity created in Home Assistant.

jamespreedy, 2023-09-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
When adding the Panic Button to Z-WaveJS UI it appears to be added successfully.

What do you observe?
The MQTT device is created in Home Assistant, and two entities are created - "battery_islow" and "battery_level".

What did you expect to happen?
No entities are created that show when the panic button has been pressed.

[image]

[image]

### Device information

Manufacturer: Philio Technology Corp
Model name: PSR03-C
Node ID in your network: 316-39-8 (0x013c-0x0008-0x0027)

https://devices.zwave-js.io/?jumpTo=0x013c:0x0008:0x0027:0.0

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.26.0.08e6a7e
zwave-js: 11.14.3
home id: 3560723676
home hex: 0xd43c54dc

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2023-09-26.log](https://github.com/zwave-js/node-zwave-js/files/12729027/zwavejs_2023-09-26.log)

Thank you in advance - apologies if I've missed something, or posted this in the wrong place. 🍺

AlCalzone, 2023-10-13:
Did you press the panic button for 5 seconds while the log was being made?
[image]

Notifications like these should be supported out of the box.

jamespreedy, 2023-10-13:
Thanks for your help @AlCalzone.

Just to be sure, I turned on debug log to file again, pressed the button a few times, and then held down the button for more than 5 seconds a couple of times too. Logs below, hope they help. If I can try anything else for you, please let me know. 😀

[zwavejs_2023-10-13.log](https://github.com/zwave-js/node-zwave-js/files/12892609/zwavejs_2023-10-13.log)

AlCalzone, 2023-10-13:
Hm, still nothing. It's node 19, correct? That one wakes up a few times - but no notification is sent. Can you share a screenshot of the "groups" tab (4th on the left)? Maybe the lifeline association is missing.

On another note: Why are you using MQTT to integrate Z-Wave JS into Home Assistant? Using the Z-Wave integration over websockets is far better supported:
https://www.home-assistant.io/integrations/zwave_js#advanced-installation-instructions
Z-Wave JS UI can be run separately, or as an addon in Home Assistant.

jamespreedy, 2023-10-13:
It is 19. Screenshot below.

[image]

I would love to use websockets and used to run the HA Addon, but I had hectic issues. Something on the zwave network was misbehaving, and it would cause the HA devices to create "bonus" entities. So for example, my electricity meter would get a luminance entity, or my motion detector would get a binary switch, or my on/off electricity plug would get a humidity sensor. I would also get loads of obviously wrong readings - house temps might be 20, 20, 21, 280, 20, so I needed up with filters and all sorts just to force a little reliability.

I split the 60 or so device network up into 3x separate zwave networks, each running off a Pi located in the 3 furthest locations within the house, and tied them back to HA via MQTT. This solved most of my issues, but I never quite got zwave working as well as everyone else seems to manage.

At the time I thought that running Z-Wave JS UI separately meant I *had* to use MQTT to connect back to HA?

jamespreedy, 2023-10-13:
Incidentally, when I used to host the whole zwave network, this panic button was fine and an entity would be created in HA. It's only since I split it up and moved to MQTT that the device doesnt have an entity of any type that catches a button press.

jamespreedy, 2023-10-13:
@AlCalzone - I just realised that wasn't what you meant at all about WS vs MQTT. I hadn't thought about it at all as I always thought I was "supposed" to use MQTT - even though I wasn't a bit fan. Have added all three Pi's using WS, screenshot of the Panic button below.

[image]

I used to use sleep state to trigger automations as it goes sleep => awake => sleep when you short press the button.

Are the entities you see in the screenshot consistent with what you'd expect to see for this device? It feels like alarmlevel or alarmtype should help me, but they stay as Unknown throughout.

Thank you again - for this and all your work in the community. 🍺

AlCalzone, 2023-10-13:
> At the time I thought that running Z-Wave JS UI separately meant I _had_ to use MQTT to connect back to HA?

No that was never the case.

> Are the entities you see in the screenshot consistent with what you'd expect to see for this device? It feels like alarmlevel or alarmtype should help me

Look at the right:
[image]
I'm not sure how to make use of that in HA though - that's a question for the community Discord.

AlCalzone, 2023-10-13:
I don't know why the notification is now appearing though. It wasn't in your previous log, so WS vs MQTT shouldn't make all the difference.

jamespreedy, 2023-10-13:
Well Discord it is. At least I can use the process I had before, and now remove my 293487293 MQTT devices. 🤣

Seriously, thank you again, really appreciate it.

closed 2023-10-13 by jamespreedy