zwave-js/zwave-js #6346: Missing device configuration: McoHome Technology Co., Ltd. MH8-FC4

tbiel5, 2023-09-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

McoHome Technology Co., Ltd. MH8-FC4 Fan Coil Thermostat

### Manufacturer ID

0x015f

### Product Type

0x0802

### Product ID

0x5101

### Firmware Version

3.2

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

[MCO Home MH8-FC4 Fan Coil Thermostat.pdf](https://github.com/zwave-js/node-zwave-js/files/12771273/MCO.Home.MH8-FC4.Fan.Coil.Thermostat.pdf)

apella12, 2023-10-11:
This should be closed with merged #6358

AlCalzone, 2023-10-11:
Oh I didn't see this wasn't linked. If you write `fixes: #<pr-number>` or `closes: #<pr-number>` in your PR description, that's done automatically.

closed 2023-10-11 by AlCalzone

tbiel5, 2023-10-11:
Thanks for your work on this @apella12 and @AlCalzone!