zwave-js/zwave-js #6356: Cannot restore a NVM Backup from Aeotec Gen5 Stick to another Aeotec gen5 stick

umtauscher, 2023-10-02:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is not described there.
- [ ] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

2.0.1

### ZwaveJS version

0.1.93

### Describe the bug

I made an NVM Backup from my currentlich running Aeotec Gen5 and tried to restore that backup to aother Stick of the same make and kind.
The restore stated with "converting ..." (for whatever reson) and after reaching 100% it stated theat it couln't find a matching converter.
The restore failed.

### To Reproduce

Just as I described above

### Expected behavior

to get a copy of my setup on a second stick

### Additional context

Homeassistant 2023.3.9  on OS10,5 using supervisor 2023.09.2

AlCalzone, 2023-10-02:
Is the firmware version on both the same?

umtauscher, 2023-10-02:
The firmware on the target stick is 1.1 SDK 6.51.10
The firmware of the source stick is 1.1 SDK 6.51.10
So yes, they are identical.

umtauscher, 2023-10-02:
additional note:
I used the backup made by zwavejs-ui and stored in on the target stick using the aeotec tool backup tool and the stick works fine. So the backup is fine, just the restore doesn't work.

AlCalzone, 2023-10-02:
Hm that's supposed to work without conversion though. Can you send me such a backup via email?

AlCalzone, 2023-10-04:
I can't find the e-mail.

umtauscher, 2023-10-04:
I hope you received it now

AlCalzone, 2023-10-04:
I'm afraid not. Do you have Discord? You could send it to me there --> https://discord.gg/HFqcyFNfWd - I have the same username there.

AlCalzone, 2023-10-12:
Ok so the issue is that Z-Wave JS UI uses the `restoreNVM` function, which checks for compatibility of the backup with the target stick. However it does not understand the format of the older backups, so it errs on the side of caution and does not allow the backup.

To restore these older backups, Z-Wave JS UI needs an option for the user to skip this check and use the `restoreNVMRaw` function instead - at the risk of restoring an incompatible backup (which can brick the controller).