zwave-js/zwave-js #6360: Multilevel Sensor CC values may be stale, refreshing - USB powered device timeouts

PeteRager, 2023-10-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
This is an Aeotec Multisensor 6 that is USB powered and is configured to send the sensors I want every 5 minutes. However, when it wakes up zwavejs decides to poll each of the multisensors individually which causes command timeouts reliably. I have 4 of these devices on two different zwave sticks. They all have the same behavior.
What do you observe?
Command Timeouts occur
What did you expect to happen?
I don't expect zwavejs to poll the device.  Or I expect a way to opt out.  Note, a number of the sensors i don't even use and hence i never report them. Perhaps that is part of the issue?
Steps to reproduce the behavior:
  1. Setup an Aeotec multisensor 6 with USB power
  2. Configure it to send a subset of sensors every 5 minutes
  3. Set the wakeup interval to 240.
  4. Observe zwavejs polling all sensors and 10-15 timeouts per day

### Device information

Manufacturer: Aeotec
Model name: zw100
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 11.14.3
`zwave-js-ui` branch: 8.26.0.08e6a7e

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Before the refresh code was added these timeouts were not as frequent 

### Attach Driver Logfile

```
2023-10-03T19:23:38.286Z DRIVER « [Node 013] [REQ] [ApplicationCommand]
                                  └─[WakeUpCCWakeUpNotification]
2023-10-03T19:23:38.286Z CNTRLR « [Node 013] received wakeup notification
[... 95 lines cut ...]
2023-10-03T19:23:40.912Z DRIVER « [REQ] [SendData]
                                    callback id:     105
                                    transmit status: OK
2023-10-03T19:23:41.924Z SERIAL « 0x010b0004000d05310505012de5                                        (13 bytes)
2023-10-03T19:23:41.925Z SERIAL » [ACK]                                                                   (0x06)
2023-10-03T19:23:41.926Z CNTRLR   [Node 013] [Multilevel Sensor] Humidity: metadata updated         [Endpoint 0]
2023-10-03T19:23:41.928Z CNTRLR   [Node 013] [~] [Multilevel Sensor] Humidity: 45 => 45             [Endpoint 0]
2023-10-03T19:23:41.929Z DRIVER « [Node 013] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Humidity
                                      scale: Percentage value
                                      value: 45
2023-10-03T19:23:41.940Z CNTRLR   [Node 013] Timed out while waiting for a response from the node (ZW0201)
2023-10-03T19:23:41.940Z DRIVER   all queues idle
2023-10-03T19:23:42.863Z SERIAL « 0x010b0004000d0531051b0100d6                                        (13 bytes)
2023-10-03T19:23:42.864Z SERIAL » [ACK]                                                                   (0x06)
2023-10-03T19:23:42.865Z CNTRLR   [Node 013] [Multilevel Sensor] Ultraviolet: metadata updated      [Endpoint 0]
2023-10-03T19:23:42.866Z CNTRLR   [Node 013] [~] [Multilevel Sensor] Ultraviolet: 0 => 0            [Endpoint 0]
2023-10-03T19:23:42.867Z DRIVER « [Node 013] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Ultraviolet
                                      scale: UV index
                                      value: 0
2023-10-03T19:23:51.491Z SERIAL « 0x010c0004000a063105012200d539                
```

PeteRager, 2023-10-04:
I see from the code that if the values have not been updated in 6 hours then it will query

https://github.com/zwave-js/node-zwave-js/blob/715a0d425c9d88a3dcd8fa1dea88ca3999698cc0/packages/cc/src/cc/MultilevelSensorCC.ts#L543

AlCalzone, 2023-10-04:
The current logic is to poll all known values every 6 hours when at least one of those hasn't been updated since. I agree this isn't ideal when you've configured to report a subset of values. Would probably be better to only poll when none of the values have been updated.

AlCalzone, 2023-10-04:
Interestingly, the sensor does respond - it's just slow.

PeteRager, 2023-10-04:
Thanks for looking and confirming that. I'll grab the logs from the other 3 sensors, from a cursory look the failure pattern is always the same.  Maybe during a poll if is trying to recalculate some of these values and that takes time?

I did reconfigure 1 of the sensors last night to send the two values that were not getting updated once an hour and that avoids the problem .  They are UV and Battery.

PeteRager, 2023-10-04:
Here is the log from yesterday, if needed, both Node 013 and Node 004 are Aoetec Multisensor 6 and exhibit the same behavior.

[zwavejs_2023-10-03.log.gz](https://github.com/zwave-js/node-zwave-js/files/12803961/zwavejs_2023-10-03.log.gz)

AlCalzone, 2023-10-04:
I guess that makes sense. If it doesn't report the values, Aeotec have probably turned off the sensor or their computation.

PeteRager, 2023-10-05:
It sounds like part of the solve maybe increasing the report timeout for this specific device in the Config file?

https://github.com/zwave-js/node-zwave-js/blob/master/docs/config-files/file-format.md#reporttimeout

If you agree, I'm happy to make the change locally and create a pull request if it solves the problem.  Seems like it should be 2s for,this device.

AlCalzone, 2023-10-05:
Can't hurt to try

closed 2023-10-13 by zwave-js-bot