zwave-js/zwave-js #6367: Incorrect default Wakeup Interval sent to device

martyn-vesternet, 2023-10-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Below log snippet from an inclusion of a battery powered device that matches https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0330/ves-zw-rem-010.json.

The device reports:

```
default interval: 0 seconds
minimum interval: 60 seconds
maximum interval: 2678400 seconds
interval steps: 60 seconds
wake up on demand supported: false
```

My (perhaps incorrect) expectation would be that the default interval is then sent to the device as configuration, instead I am seeing a value of 60 seconds being sent:

```
wake-up interval: 60 seconds
controller node id: 1
```

This doesn't seem correct, since it then means the device wakes up every minute and depletes the battery within a few days.

I can see that the device reports that its *minimum* and *steps* intervals are 60 seconds, but don't believe that it should be using those to set the default?

In the absence of a specific configuration for Wakeup Interval within ves-zw-rem-010.json, should it not use the default that was reported from the device during interview?

I can obviously update the ves-zw-rem-010.json file (and other battery device files) to set a specific Wakeup Interval, but this does seem to be a bug if the default reported from the device should have been used if no specific value was set?

Checking WakeUpCC.ts at https://github.com/zwave-js/node-zwave-js/blob/2da6d9e1c6f3253df0d4dc95e2b8b62a6ca48175/packages/cc/src/cc/WakeUpCC.ts#L312C1-L312C1 it seems that the desiredInterval is clamped between minInterval and maxInterval, so it does seem to be the intended behaviour.

If so then I guess it's not a bug as such, albeit that setting to 0 is a common way to disable WakeUp while still allowing a different minimum value to prevent users doing something crazy like setting WakeUp to 1 second.

The above change to WakeUpCC.ts seems to be part of https://github.com/zwave-js/node-zwave-js/pull/5936 which possibly explains why this issue has only recently surfaced in the past couple of months. 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 12.0.1
`zwave-js-ui` branch: 9.0.3

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Prior to https://github.com/zwave-js/node-zwave-js/pull/5936

### Attach Driver Logfile

```
2023-10-05T09:41:50.730Z CNTRLR [Node 021] Embedded device config loaded
2023-10-05T09:41:50.735Z CNTRLR [Node 021] Interviewing Wake Up...
2023-10-05T09:41:50.736Z CNTRLR » [Node 021] retrieving wakeup capabilities from the device...
2023-10-05T09:41:50.744Z DRIVER all queues busy
2023-10-05T09:41:50.750Z DRIVER » [Node 021] [REQ] [SendDataBridge]
│ source node id: 1
│ transmit options: 0x25
│ callback id: 48
└─[Security2CCMessageEncapsulation]
│ sequence number: 23
│ security class: S2_Unauthenticated
└─[WakeUpCCIntervalCapabilitiesGet]
2023-10-05T09:41:50.763Z DRIVER « [RES] [SendDataBridge]
was sent: true
2023-10-05T09:41:50.870Z DRIVER « [REQ] [SendDataBridge]
callback id: 48
transmit status: OK, took 110 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
routing scheme: LWR
ACK RSSI: -14 dBm
ACK channel no.: 0
TX channel no.: 0
2023-10-05 10:41:50.902 INFO Z-WAVE: [Node 021] Metadata updated: 132-0-wakeUpInterval
2023-10-05T09:41:50.907Z DRIVER « [Node 021] [REQ] [BridgeApplicationCommand]
│ RSSI: -14 dBm
└─[Security2CCMessageEncapsulation]
│ sequence number: 29
│ security class: S2_Unauthenticated
└─[WakeUpCCIntervalCapabilitiesReport]
default interval: 0 seconds
minimum interval: 60 seconds
maximum interval: 2678400 seconds
interval steps: 60 seconds
wake up on demand supported: false
2023-10-05T09:41:50.909Z CNTRLR « [Node 021] received wakeup capabilities:
default wakeup interval: 0 seconds
minimum wakeup interval: 60 seconds
maximum wakeup interval: 2678400 seconds
wakeup interval steps: 60 seconds
wakeup on demand supported: false
2023-10-05T09:41:50.910Z CNTRLR » [Node 021] retrieving wakeup interval from the device...
2023-10-05T09:41:50.927Z DRIVER » [Node 021] [REQ] [SendDataBridge]
│ source node id: 1
│ transmit options: 0x25
│ callback id: 49
└─[Security2CCMessageEncapsulation]
│ sequence number: 24
│ security class: S2_Unauthenticated
└─[WakeUpCCIntervalGet]
2023-10-05T09:41:50.947Z DRIVER « [RES] [SendDataBridge]
was sent: true
2023-10-05T09:41:51.013Z DRIVER « [REQ] [SendDataBridge]
callback id: 49
transmit status: OK, took 70 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
routing scheme: LWR
ACK RSSI: -14 dBm
ACK channel no.: 0
TX channel no.: 0
2023-10-05 10:41:51.030 INFO Z-WAVE: [Node 021] Value added: 132-0-wakeUpInterval => 0
2023-10-05 10:41:51.035 INFO Z-WAVE: [Node 021] Metadata updated: 132-0-controllerNodeId
2023-10-05 10:41:51.038 INFO Z-WAVE: [Node 021] Value added: 132-0-controllerNodeId => 255
2023-10-05T09:41:51.041Z DRIVER « [Node 021] [REQ] [BridgeApplicationCommand]
│ RSSI: -14 dBm
└─[Security2CCMessageEncapsulation]
│ sequence number: 30
│ security class: S2_Unauthenticated
└─[WakeUpCCIntervalReport]
wake-up interval: 0 seconds
controller node id: 255
2023-10-05T09:41:51.043Z CNTRLR « [Node 021] received wakeup configuration:
wakeup interval: 0 seconds
controller node: 255
2023-10-05T09:41:51.044Z CNTRLR » [Node 021] configuring wakeup destination node
2023-10-05T09:41:51.065Z DRIVER » [Node 021] [REQ] [SendDataBridge]
│ source node id: 1
│ transmit options: 0x25
│ callback id: 50
└─[Security2CCMessageEncapsulation]
│ sequence number: 25
│ security class: S2_Unauthenticated
└─[SupervisionCCGet]
│ session id: 11
│ request updates: true
└─[WakeUpCCIntervalSet]
wake-up interval: 60 seconds
controller node id: 1
2023-10-05T09:41:51.096Z DRIVER « [RES] [SendDataBridge]
was sent: true
2023-10-05T09:41:51.136Z DRIVER « [REQ] [SendDataBridge]
callback id: 50
transmit status: OK, took 60 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
routing scheme: LWR
ACK RSSI: -14 dBm
ACK channel no.: 0
TX channel no.: 0
2023-10-05T09:41:51.172Z DRIVER « [Node 021] [REQ] [BridgeApplicationCommand]
│ RSSI: -14 dBm
└─[Security2CCMessageEncapsulation]
│ sequence number: 31
│ security class: S2_Unauthenticated
└─[SupervisionCCReport]
session id: 11
more updates follow: false
status: NoSupport
duration: 0s
2023-10-05 10:41:51.179 INFO Z-WAVE: [Node 021] Value added: 108-0-ccSupported-132 => false
2023-10-05T09:41:51.191Z DRIVER » [Node 021] [REQ] [SendDataBridge]
│ source node id: 1
│ transmit options: 0x25
│ callback id: 51
└─[Security2CCMessageEncapsulation]
│ sequence number: 26
│ security class: S2_Unauthenticated
└─[WakeUpCCIntervalSet]
wake-up interval: 60 seconds
controller node id: 1
2023-10-05T09:41:51.201Z DRIVER « [RES] [SendDataBridge]
was sent: true
2023-10-05T09:41:51.279Z DRIVER « [REQ] [SendDataBridge]
callback id: 51
transmit status: OK, took 70 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
routing scheme: LWR
ACK RSSI: -14 dBm
ACK channel no.: 0
TX channel no.: 0
2023-10-05 10:41:51.794 INFO Z-WAVE: [Node 021] Value updated: 132-0-controllerNodeId 255 => 1
2023-10-05T09:41:51.800Z CNTRLR [Node 021] wakeup destination node changed!
```

martyn-vesternet, 2023-10-05:
[z-wave-js_log.txt](https://github.com/zwave-js/node-zwave-js/files/12818053/z-wave-js_log.txt)

AlCalzone, 2023-10-06:
I'll have to double check, but if I remember correctly, this was done because of a requirement from the certification test suite. There's one situation where the reported default interval lies outside of the min/max range, and Z-Wave JS trying to preserve that would cause the test to fail.
I'm not sure though if that default was 0 or not.

Going by this though...
```
default interval: 0 seconds
minimum interval: 60 seconds
maximum interval: 2678400 seconds
interval steps: 60 seconds
wake up on demand supported: false
```
...why isn't the minimum interval 0 here? Having a step size of 60 would make sure that the next higher legal value is 60 anyways.

martyn-vesternet, 2023-10-06:
>>>>why isn't the minimum interval 0 here
>>>>Having a step size of 60 would make sure that the next higher legal value is 60 anyways.

I see what you mean, the device should reject less than 60 due to the step size while still allowing 0 to be set.

I did just test that by attempting to set to 32 and it appears to accept it but on read back it returns 60, so presume it takes anything from 1 - 59 as 60. It's the same at the top end, if I set 2678402 then on read back it returns 2678400.

From a device perspective I'm not sure what the correct behaviour should be - should anything outside the minimum / maximum be rejected with an error? 

If so then I guess Sunricher may have interpreted that incorrectly as it looks like they're returning the WakeUp minimum that device actually works with, rather than the lowest value that will be accepted.

So perhaps it's a device bug after all then, unfortunately it does mean though that this likely impacts all Sunricher based battery powered devices :-(

AlCalzone, 2023-10-11:
I can't find anymore which CTT test was failing before this change, but it was a weird situation where the wakeup interval returned by the device itself was outside of the allowed range.

In any case, I think it makes sense to try and preserve the configured interval if it's outside the allowed range, but 0.

PR #6387 (2023-10-11): fix: preserve wakeup interval of 0 (never wake up), even if device claims this is not valid
fixes: #6367
closed 2023-10-11 by zwave-js-bot