zwave-js/zwave-js #6376: Z-Wave JS - wrong discovery of TZ65D

BaconStivvy, 2023-10-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

when trying to add a device called TZ65D it is being detected as a TZ65S losing functionality of a dual switch
[zwave_js-6ebd3030d04489d308a51b9fc7acf83a-Single Paddle Wall Dimmer-9ecff1430f42fe52337a47ce713956d7.txt.txt](https://github.com/zwave-js/node-zwave-js/files/12832072/zwave_js-6ebd3030d04489d308a51b9fc7acf83a-Single.Paddle.Wall.Dimmer-9ecff1430f42fe52337a47ce713956d7.txt.txt)

it is a known issue already reported here [https://github.com/home-assistant/core/issues/83283](url) and 
here [https://github.com/home-assistant/core/issues/101264#issuecomment-1750755317](url)

### Device information

Manufacturer: TKB Home
Model name: TZ-65D
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave_js-6ebd3030d04489d308a51b9fc7acf83a-Single Paddle Wall Dimmer-9ecff1430f42fe52337a47ce713956d7.txt.txt](https://github.com/zwave-js/node-zwave-js/files/12832167/zwave_js-6ebd3030d04489d308a51b9fc7acf83a-Single.Paddle.Wall.Dimmer-9ecff1430f42fe52337a47ce713956d7.txt.txt)

BaconStivvy, 2023-10-06:
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12832429/zwave_js.log)

apella12, 2023-10-07:
As noted in the HA threads the device advertises itself with the TYPE:ID aligned with the single paddle model.  

I'm new to ZwaveJS, do not have the device, and have never used HA, but maybe there is a manual way.  I do see from the log that the device advertises 4 association groups.  These are what the right-side paddle controls. At least in Zwave-js-ui you can go to the device and go the Groups menu. Are there four shown? If not try to ADD a group 2 (from root) and put some node that you want to control using the right-side paddle in it.  Then on the Config section send a custom parameter (14 1 1) to activate group 2.  See if that works.  It is a longshot

AlCalzone, 2023-10-09:
Looks like we have to expose the Basic Set "events" from the 2nd paddle like [we did for the TZ66D](https://github.com/zwave-js/node-zwave-js/issues/4841#issuecomment-1281307325).

AlCalzone, 2023-10-09:
@BaconStivvy The next release should fix this, but you will still need to re-interview the device then and add the association group as @apella12 outlined.

closed 2023-10-09 by zwave-js-bot

BaconStivvy, 2023-10-09:
that will be great thanks. i will wait for next release and let you know here :)

BaconStivvy, 2023-10-15:
Hi, I confirm that secondary switch is being detected in logs after i re included device.

Is group association possible in z-wave js? Still very green in yaml and have no idea what to do. 

Or at least is it possible to add as an entity then I can configure second switch to control in automation? Any help would be much appreciated

AlCalzone, 2023-10-16:
Association groups can only be managed in the **Z-Wave JS UI** addon, not the **Z-Wave JS** addon. You should have a "Basic event" entity or something like this now after re-interviewing. The Home Assistant Discord may be better to ask for help with this.

BaconStivvy, 2023-10-16:
thanks for your help

PR #6381 (2023-10-09): fix(config): treat Basic Set as events for TKB TZ35/55 S/D
fixes: #6376