zwave-js/zwave-js #6392: Add resilience for corrupted ManufacturerSpecificCCReport

PeteRager, 2023-10-11:
**Is your feature request related to a problem? Please describe.**
Yes, when a battery powered device wakes-up, zwavejs queries VersionCCReport and ManufacturerSpecificCCReport. Sometimes this data is corrupted due to limitations on the zwave checksum. When this happens the device is reported as unknown and it causes upstream breakages in HA.  Because the likelihood of this data actually changing is very low - especially the manufacturer id, product type and product id - and the level of existing problems that have been reported - a better solution is needed.

**Describe the solution you'd like**
When this situation is encountered, the system should re-verify the data by requesting it again.  And only if the two responses match should it be processed.  This could also be extended to address corrupted reports for CC classes that are not part of the registered device.

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here. If you want to propose an API, share your thoughts how it could 

Here is a log demonstrating the issue.  634 and 28672 are the correct answers.  It is a zooz zse41 door sensor.

From the zwavejs DB
```
	"manufacturer": "Zooz",
	"manufacturerId": "0x027a",
	"label": "ZSE41",
	"description": "Open/Close XS Sensor",
	"devices": [
		{
			"productType": "0x7000",
			"productId": "0xe001",
			"zwaveAllianceId": 4276
		}
	],
```

It's also not clear why it seems to request the data multiple times and some of the same answers are received.  If it truly was RF corruption I'd expect 2867 not to be reported multiple times?

```
2023-10-11T00:37:38.349Z SERIAL « 0x010e0004001f087205027a7000e0017c                                  (16 bytes)
2023-10-11T00:37:38.350Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:38.351Z CNTRLR   [Node 031] [~] [Manufacturer Specific] manufacturerId: 634 => 634 [Endpoint 0]
2023-10-11T00:37:38.352Z CNTRLR   [Node 031] [~] [Manufacturer Specific] productType: 28672 => 2867 [Endpoint 0]
                                  2
2023-10-11T00:37:38.353Z CNTRLR   [Node 031] [~] [Manufacturer Specific] productId: 57345 => 57345  [Endpoint 0]
2023-10-11T00:37:38.354Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[ManufacturerSpecificCCReport]
                                      manufacturer id: 0x027a
                                      product type:    0x7000
                                      product id:      0xe001
2023-10-11T00:37:38.355Z CNTRLR « [Node 031] received expected response prematurely, remembering it...
2023-10-11T00:37:39.266Z SERIAL « 0x01070013aa00010040                                                 (9 bytes)
2023-10-11T00:37:39.267Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:39.267Z DRIVER « [REQ] [SendData]
                                    callback id:     170
                                    transmit status: OK
2023-10-11T00:37:39.272Z SERIAL » 0x010900131f02861125abe1                                            (11 bytes)
2023-10-11T00:37:39.272Z DRIVER » [Node 031] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      171
                                  └─[VersionCCGet]
2023-10-11T00:37:39.273Z SERIAL « [ACK]                                                                   (0x06)
2023-10-11T00:37:39.280Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-10-11T00:37:39.280Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:39.281Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-10-11T00:37:40.209Z SERIAL « 0x01070013ab00005d1d                                                 (9 bytes)
2023-10-11T00:37:40.209Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:40.210Z DRIVER « [REQ] [SendData]
                                    callback id:     171
                                    transmit status: OK
2023-10-11T00:37:40.365Z SERIAL « 0x010e0004001f087205027a7000e0017c                                  (16 bytes)
2023-10-11T00:37:40.365Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:40.366Z CNTRLR   [Node 031] [~] [Manufacturer Specific] manufacturerId: 634 => 634 [Endpoint 0]
2023-10-11T00:37:40.367Z CNTRLR   [Node 031] [~] [Manufacturer Specific] productType: 28672 => 2867 [Endpoint 0]
                                  2
2023-10-11T00:37:40.368Z CNTRLR   [Node 031] [~] [Manufacturer Specific] productId: 57345 => 57345  [Endpoint 0]
2023-10-11T00:37:40.369Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
[... 27 lines cut ...]
                                      firmware versions: 1.30
                                      hardware version:  1
2023-10-11T00:37:47.211Z SERIAL « 0x010e0004001f08720502060c00e0017c                                  (16 bytes)
2023-10-11T00:37:47.212Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:47.213Z CNTRLR   [Node 031] [~] [Manufacturer Specific] manufacturerId: 634 => 518 [Endpoint 0]
2023-10-11T00:37:47.214Z CNTRLR   [Node 031] [~] [Manufacturer Specific] productType: 28672 => 3072 [Endpoint 0]
2023-10-11T00:37:47.215Z CNTRLR   [Node 031] [~] [Manufacturer Specific] productId: 57345 => 57345  [Endpoint 0]
2023-10-11T00:37:47.216Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[ManufacturerSpecificCCReport]
                                      manufacturer id: 0x0206
                                      product type:    0x0c00
                                      product id:      0xe001
2023-10-11T00:37:47.307Z SERIAL « 0x010f0004001f09861203070d011e010061                                (17 bytes)
2023-10-11T00:37:47.308Z SERIAL » [ACK]                                                                   (0x06)
2023-10-11T00:37:47.308Z CNTRLR   [Node 031] [~] [Version] libraryType: 3 => 3                      [Endpoint 0]
2023-10-11T00:37:47.309Z CNTRLR   [Node 031] [~] [Version] protocolVersion: "7.13" => "7.13"        [Endpoint 0]
2023-10-11T00:37:47.311Z CNTRLR   [Node 031] [~] [Version] firmwareVersions: 1.30 => 1.30           [Endpoint 0]
2023-10-11T00:37:47.312Z CNTRLR   [Node 031] [~] [Version] hardwareVersion: 1 => 1                  [Endpoint 0]
2023-10-11T00:37:47.313Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[VersionCCReport]
                                      library type:      Enhanced Slave
                                      protocol version:  7.13
                                      firmware versions: 1.30
                                      hardware version:  1
```

PeteRager, 2023-10-11:
zwave-js-ui: 8.14.2.c234053
zwave-js: 10.16.0

PeteRager, 2023-10-11:
The should avoid this from happening as frequently.

In the log it looked like it queried it multiple times and got the same bad value back.  Am I reading that correct?  If so, it seems impropable that RF interference would cause the exact same error?

AlCalzone, 2023-10-11:
> queried it multiple times and got the same bad value back

The repeated ones are the correct identifier though (the line is wrapping). The last one seems wrong.

Anyways, you need to update, the mentioned change was done specifically to prevent this.

closed 2023-10-11 by AlCalzone