zwave-js/zwave-js #6393: Swiid / CBC switch dead after a while

weha, 2023-10-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown cause, but it only happens with ZwaveJS, it worked fine for years on Fibaro router

What do you observe?
Everything works find but after a while the device is marked as dead and needs to be re-added to the network.

What did you expect to happen?
No need to re-add it after n time

I have had this devices for many years, one at my place and one at my parents.
Both houses ran on Fibaro routers, I switched mine and after a while this device stopped working and I needed to re-add it.
A few years later I changed to zwavejs at my parents house, and now they have the same issue with their device.
This leads me to believe something goes wrong on the zwavejs end.
I don't have debug log of it as it is impossible to know when it actually happens.
I'll re-add the device and enable debug logging but I would like to know if there is anything else I can do or should know?
This can take a few months so I'd rather know beforehand so I don't need to wait another few months.
Does it help to have a debug when I re-add it?

### Device information

Manufacturer: Swiid
Model name: SWI_ZCS1

But zwave-js identifies it as this device: https://devices.zwave-js.io/?jumpTo=0x0166:0x0100:0x0100:0.0
potàto potáto I guess?

Node ID in your network: 13

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 9.1.2.a13a720
zwave-js: 12.0.4

My zwave stick is a Aeotec Gen5, 1.1 version I think it was back then.
My parents zwave stick as Aeotec Z-stick (not sure which version, it's pretty old though)

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### Attach Driver Logfile

[zwavejs_2023-10-12.log](https://github.com/zwave-js/node-zwave-js/files/12878314/zwavejs_2023-10-12.log)

AlCalzone, 2023-10-12:
> transmit status: NoAck

This is the stick telling us that it didn't get an acknowledgement from the device that it got the command, after several attempts.

Sending commands in Z-Wave roughly works like this:
- Z-Wave JS to stick: "Here, send this"
- ...
- stick to Z-Wave JS: "I sent it, but the node didn't respond"

This is an oversimplification, but as you can see, Z-Wave JS has no influence on what happens after telling the stick to send something. That part of communication depends entirely on the stick's hardware, software [and location](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting), the device's hardware, software and location, as well as the physical environment (sources of RF noise and interferences, etc...)
I get similar arguments often from users switching from OpenZWave, but that just didn't expose communication issues as clearly as Z-Wave JS does, so users often didn't notice that they had bad connectivity.

Not sure what you changed when switching to Z-Wave JS, but there might be a few factors that could cause this:
- different controller (stick) with different RF performance/range?
- different location of the controller?
- added or removed other devices?
- bought a new microwave?

Being dead/unreachable is often just a temporary state. If you really need to reset the device and include it again, and things pushing a button on the device does not bring it back, this does hint at a problem with the device itself.

weha, 2023-10-12:
Well I guess I did not even consider it could be an issue with the stick in relation to the device.
Like I said, the device works fine with Fibaro routers, but I guess it does not like the Aeotec sticks.
it is definitely in a different location, there was surely a direct connection before but know perhaps with a hop.

So I suppose you can close this, it's probably indeed a hardware issue, unless there is anything I can do about it?
Should I try and get the debug when it goes dead or will that not be interesting?

AlCalzone, 2023-10-12:
> Should I try and get the debug when it goes dead or will that not be interesting?

It probably won't give us any additional information.

You can try following these steps/instructions to try and improve the situation:
1. https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
2. https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength