zwave-js/zwave-js #6399: ZST39 is probably corrupting the ACK after a soft reset

jtbraun, 2023-10-15:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The issue of a soft reset to a ZST39 during the startup phase.

What do you observe?
Soft reset is issued, and an 0x86 byte is immediately received and discarded. Then the SerialAPIStarted message comes in, and later the driver complains that an ACK was never received.

What did you expect to happen?
ACK would be received.

Steps to reproduce the behavior:
  1. Get a Zooz ZST39 800-series stick
  2. Update to latest (7.19 I think?)
  3. Run the `test/run.ts` aka "Debug locally" test in vscode
  4. See screenshotted below.

[image]

Modifying `packages/serial/src/parsers/SerialAPIParser.ts` to also accept 0x86 as "ACK" gets past the error.

NOTE: I've reported the issue to Zooz, if I hear back from them I'll let you know. In the meantime, there probably needs to be a quirk flag or something that is set in the serialAPIParser specifically when softreset is waiting on an ACK, that treats a discarded byte as the missing ACK. :-/

### Device information

Manufacturer: ZOOZ
Model name: ZST39
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: master (9818c58c9cb0e4451a8856b8d69330c4a03fbd03)
`node-zwave-js` branch: master
`zwave-js-ui` branch: n/a

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I changed the COM port to COM3 instead of COM5 in run.ts

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

n/a

### Attach Driver Logfile

[zwave.log](https://github.com/zwave-js/node-zwave-js/files/12909655/zwave.log)

AlCalzone, 2023-10-16:
Hmm, I've seen something similar with the "official" USB controllers too: https://community.silabs.com/s/question/0D58Y00009pca5bSAA/bug-in-firmware-7191-softreset-is-no-longer-answered-with-ack?language=en_US

The 7.19 firmware is just broken :(

AlCalzone, 2023-10-16:
Correct. I have a semi-direct contact to Silabs and made them aware of it again. Hopefully this gains some traction now.

AlCalzone, 2023-10-17:
Fix is up here: https://github.com/zwave-js/node-zwave-js/pull/6409
I have seen different values (`c6`, `e6`) on other controllers, so my strategy for now is to ignore the high nibble of the ACK byte completely, but only once.

closed 2023-10-17 by zwave-js-bot