zwave-js/zwave-js #6405: Thermostat Mode getSupportedModes override not working correctly

poelstra, 2023-10-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**

My Eurotronic Spirit thermostatic valve (well, a clone brand) supports a "Manufacturer Specific" mode as one of the Thermostat Modes. There's already a warning added to it (https://github.com/zwave-js/node-zwave-js/pull/4144) but I'd thought to use the now-available compat mechanism to just fix it.

I copied the existing spirit.json to a new custom config (in store/config/spirit.json in ZWave-JS UI), removed the existing warning and added the following:
```json
	"compat": {
		"overrideQueries": {
			// The device does not report "Manufacturer specific" as a supported thermostat mode
			"Thermostat Mode": [
				{
					"method": "getSupportedModes",
					"result": [0, 1, 11, 15, 31] // Off / Heat / Energy heat / Full power / Manufacturer specific
				}
			]
		}
	}
```

I then pressed Save in UI's settings to reload, clicked the re-interview button for my device.

**What do you observe?**

In the logs, I get (as expected):

```
2023-10-15T16:23:39.259Z CNTRLR   [Node 007] Interviewing Thermostat Mode...
2023-10-15T16:23:39.259Z CNTRLR » [Node 007] querying supported thermostat modes...
2023-10-15T16:23:39.259Z CNTRLR   [Node 007] API call getSupportedModes for Thermostat Mode CC overridden by a c
                                  ompat flag.
2023-10-15T16:23:39.259Z CNTRLR « [Node 007] received supported thermostat modes:
                                  · Off
                                  · Heat
                                  · Energy heat
                                  · Full power
                                  · Manufacturer specific
```

But in the UI I then get all modes listed (including Manufacturer specific):

[image]

**What did you expect to happen?**

I expected to see just the modes mentioned in the config and log.

I also looked at https://github.com/zwave-js/node-zwave-js/blob/15ed348f4549951eecda1e22a63b8af2f98b76fb/packages/config/config/devices/0x0115/zme_ft.json#L122 for inspiration of a similar issue in another device.
That one includes additional `persistValues` and `extendMetadata` fields, which seem redundant. I did try them though, as follows:

```json
	"compat": {
		"overrideQueries": {
			// The device does not report "Manufacturer specific" as a supported thermostat mode
			"Thermostat Mode": [
				{
					"method": "getSupportedModes",
					"result": [0, 1, 11, 15, 31], // Off / Heat / Energy heat / Full power / Manufacturer specific
					"persistValues": {
						"supportedModes": [0, 1, 11, 15, 31]
					},
					"extendMetadata": {
						"thermostatMode": {
							"states": {
								"Off": 0,
								"Heat": 1,
								"Energy heat": 11,
								"Full power": 15,
								"Manufacturer specific": 31
							}
						}
					}
				}
			]
		}
	}
```

But that gives in the UI (after re-interview):

[image]

(So I suspect that this doesn't work correctly in the zme_ft device either, but that's probably another bug then?)

I'd like to understand whether this should indeed be the correct direction, if so, I can provide more debugging if needed.

### Device information

Manufacturer: Eurotronics
Model name: Spirit
Node ID in your network: 7

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Logs of the attempt including the persistValue/extendMetadata:

[zwavejs_overriden_modes.log](https://github.com/zwave-js/node-zwave-js/files/12921718/zwavejs_overriden_modes.log)

I can recreate the one without those two too, if necessary.

AlCalzone, 2023-10-17:
The `extendMetadata` is indeed the intended way due to how this compat flag works internally. For some CCs, Z-Wave JS persists a command's values and metadata in the value DB when the actual command is received. This is the case for `Thermostat Mode CC`. For others it populates the metadata during the interview based on what's returned by the API call used to query the device.

As the compat flag completely skips any communication and simply fakes the API call by returning the defined value, no metadata is persisted automatically, so this has to be faked too as part of the compat flag.
There seems to be an issue with the implementation though if the UI looks like that.

AlCalzone, 2023-10-18:
@poelstra the metadata was just wrong: https://github.com/zwave-js/node-zwave-js/pull/6420/files

poelstra, 2023-10-18:
> [Tracking afbeelding][https://github.com/notifications/beacon/AAH2MXCN3HNRLJYKHTR65PLX7637BA5CNFSM6AAAAAA6CZFG42WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTJMVDIG.gif]

Thnx for the blazing fast response ;)
I'll try it out tomorrow.

AlCalzone, 2023-10-19:
I also have a fix up - should land next release

closed 2023-10-19 by zwave-js-bot

poelstra, 2023-10-19:
Works like a charm, thanks!