zwave-js/zwave-js #6407: Z-wave Receiving Commands but Not Sending (800-series controller)

asayler, 2023-10-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Unknown

What do you observe?

After running correctly for some time (at least days, if not longer), Z-wave JS stops successfully sending z-wave commands, even though it continues to receive commands correctly. Thus (most) devices still update, but controlling devices is not possible. Fully rebooting the hardware seems to remedy the issue.

It's possible this relates to #6260, but filing a new bug since that oen is resolved. 

What did you expect to happen?

Z-wave commands are successfully sent to devices.

Steps to reproduce the behavior:

Unknown

### Device information

I am running:

zwave-js-ui: 9.1.2
zwave-js: 12.0.4

My controller is a Zooz ZAC93 running the (problematic) 1.20 (SDK 7.19.3) firmware.

I am running Zwave JS UI via the 2.1.2 HA plugin under HA 2023.10.3 using HAOS 11.0.

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

N/A

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I did recently upgrade HAOS to 11.0, but it's not clear if that's associated with this issue.

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2023-10-16.log](https://github.com/zwave-js/node-zwave-js/files/12923673/zwavejs_2023-10-16.log)

asayler, 2023-10-17:
I'm not 100% sure when the issue started in the logs, but things seemed to be working correctly prior to `2023-10-17T00:25:17.600Z` or so since I know the automated Z-wave lights (Node 180) that trigger around then came on correctly.

AlCalzone, 2023-10-17:
What seems to happen is the same issue you already linked to. The implemented recovery mechanism also works.

But your power meters and multilevel sensors that tried to send a report to the controller during the time it was unresponsive went in full-on panic mode and started repeating their reports over and over via different routes, to the point that
1. the recovery mechanism needed 3 attempts before being able to restart the controller (as there was so much traffic incoming)
2. the network is so flooded with commands that even after restarting, the controller is unable to reach some nodes (or their acknowledgement doesn't find the way back to the controller)

Unfortunately this is nothing we can work around. I recommend going through [this guide](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=optimizing-the-reporting-configuration) and applying it to reduce the strain on the network and the likelyhood of this happening until the firmware update drops.

asayler, 2023-10-18:
Thanks, @AlCalzone. I'll see if I can reduce the network load some and continue to wait for the permanent firmware fix.