zwave-js/zwave-js #6418: unkown manufacturer

zeev-mindali, 2023-10-18:
Dear all,

node 18 and node 13 reporting that it not recognise the manufacturer
[image]

but in pc controller, i get the manufacturer : 0x015f

here are the logs files:
[z-ui_2023-10-18.log](https://github.com/zwave-js/zwave-js-ui/files/12981694/z-ui_2023-10-18.log)

is it a bug from zwave-js or as usual the McoHome product

zeev-mindali, 2023-10-18:
made re-interview with no success,
deleted old log and make reboot to the device to get a clean log.
[zwavejs_2023-10-18.log](https://github.com/zwave-js/node-zwave-js/files/12998708/zwavejs_2023-10-18.log)

AlCalzone, 2023-10-18:
I don't see an interview attempt in that log. Is this a battery-powered device? If so it needs to be woken up manually, usually by pressing a button on the device.

zeev-mindali, 2023-10-18:
it's a glass switch 1 gang
i attching latest driver log again.

[zwavejs_2023-10-18.log](https://github.com/zwave-js/node-zwave-js/files/12999488/zwavejs_2023-10-18.log)

AlCalzone, 2023-10-18:
I see an interview for node 24, but we don't have a config file for that one either - so no labels.

zeev-mindali, 2023-10-18:
i can make the config file, but it's doesn't give me manufactorer, product id, product type
if i had those numbers, i could write one

AlCalzone, 2023-10-18:
Your log did include them, so they should also be available via the UI

zeev-mindali, 2023-10-18:
if it was simple :(

the ui don't give me any information for building the config file
[image]

maybe i can get the manufactorer id, product id and product type somehow from the log?

zeev-mindali, 2023-10-18:
[image]

AlCalzone, 2023-10-18:
Ok I take that back. I still haven't seen a log of node 23 being re-interviewed. Only node 24, but we don't have a config file for that.

AlCalzone, 2023-10-18:
Ok Node 23 is not responding at all to any queries. I think you may have to exclude it, factory reset it and include it again.

apella12, 2023-10-18:
There is this in your log file.
```
2023-10-18T07:48:00.555Z DRIVER « [Node 024] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -57 dBm
                                  └─[ManufacturerSpecificCCReport]
                                      manufacturer id: 0x015f
                                      product type:    0x2221
                                      product id:      0x5106
```

zeev-mindali, 2023-10-18:
Factory reset solved the issue :)

you are the best

closed 2023-10-18 by zeev-mindali

AlCalzone, 2023-10-18:
We don't have a config file for node 24 though. I'd appreciate it if you could at least share the manual for it. Or write a config file if you have the time.