zwave-js/zwave-js #6428: Support Learn Mode to become a secondary controller

AlCalzone, 2023-10-19:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/6427

<div type='discussions-op-text'>

<sup>Originally posted by **@thomaswitt** October 19, 2023</sup>
### Description

I bought an Aeotec Z-Stick 7 for my RPi 4 and hoped it could be used as a secondary controller for my Homey Pro system (as the Homey Aeotec App supports the Z-Stick Gen 5 officially). There's even some [official documentation](https://aeotec.freshdesk.com/support/solutions/articles/6000263528-z-stick-gen-5-as-secondary-controller-in-homey) on the Aeotec page for that. I wanted to gain three functionalities by using Z-Wave JS UI with that:

1. Seamless MQTT integration - so that all actions of all my devices can be broadcasted to my MQTT broker.
2. Firmware Updates for Z Wave Devices (as Homey doesn't support Firmware Updates)
3. Getting easy API access to send commands to Z Wave Devices (e.g. turning on/off a light via an API call easily)

Unfortunately, Z-Wave JS UI - as I understand - doesn't support learn mode at the moment and therefore I can't include it as a device/secondary controller in my existing network.

### Do you have a solution in mind?

I would suggest to add support for Z-Wave UI JS for being a secondary controller.

### Describe alternatives you've considered

There are non as I don't want to drop Homey as my primary system.

### Additional context

_No response_</div>

johnderose, 2024-06-12:
Secondary Controller capability would be amazing. I'm migrating from HomeSeer4 with four Z-Net controllers (all on same network; it's a big house) and partitioning my Z-Wave network would be sub-optimal. This part of HomeSeer's solution has worked great for years. I bet we can do it! Sign me up as a tester...

AlCalzone, 2024-06-12:
To be completely transparent: this won't happen in the near future. Learn Mode support adds a bunch of extra requirements for certification which I don't want to deal with at the moment.

Maybe next year though.

closed 2024-09-06 by zwave-js-bot

37Rb, 2025-01-07:
I would love to see secondary controller support as well. I'm working on a project that would benefit from connecting to alarm systems like the Qolsys IQ Panel as a secondary controller. I'd be willing to help build the feature if you could point me in the right direction, or contribute financially to get this supported.

ismell, 2025-01-30:
I have a similar use case as https://github.com/zwave-js/node-zwave-js/issues/6428#issuecomment-2575752458. I have a Vivnt security panel with a bunch of z-wave sensors. I would like to join a secondary controller to the network so that I can get door/window open/close notifications into HA. Is https://github.com/zwave-js/node-zwave-js/issues/1070 the correct issue to track the remaining work, or should this issue be reopened?

Thanks for all the work you put into this project!

PR #7135 (2024-09-06): feat: support learn mode
This PR adds support for joining another network using S2 and S0 and being a secondary controller

fixes: #6428

enables https://github.com/zwave-js/node-zwave-js/issues/7096
enables #1070