zwave-js/zwave-js #6440: Missing device configuration: Ring Alarm Panic Button 2nd Gen

Deimos-bg, 2023-10-22:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Ring Alarm Panic Button 2nd Gen

### Manufacturer ID

0x0346

### Product Type

0x0801

### Product ID

0x0401

### Do you have a manual?

I found it on the official Ring website. In addition to the upload, here is the original link: https://support.ring.com/hc/en-us/articles/14765129591444-Ring-Alarm-Panic-Button-2nd-gen-Z-Wave-Technical-Manual

Attachment:
[Ring Alarm Panic Button Gen 2 Zwave.pdf](https://github.com/zwave-js/node-zwave-js/files/13064002/Ring.Alarm.Panic.Button.Gen.2.Zwave.pdf)

apella12, 2023-10-24:
@Deimos-bg Is there a model number for the device on the box? I looked at Panic button gen_1 and it was 4AP1S90EN0.  Trying to avoid duplication in the device display. Thanks !

Deimos-bg, 2023-10-24:
@apella12 
Hi, 
Unfortunately, I don't have the device box anymore.... but luckily, it seems like the model number is written on the device itself.

It states: 5F56E9
Please take a look at the attachment for reference.

I'm open to help however I can with this. I just don't know how to do this configuration myself.

[image]

apella12, 2023-10-24:
@Deimos-bg Thanks.  I'll go with that pending review.

For understanding, you can go to the pull request and look at the one file changed (added).  Basically it is a structured JSON file. I use a fork of the zwave-js and edit in VSC.  VSC will show the errors, if you make them. If you have the device (I don't) you can put the draft in the store/config folder in ZUI (I don't know about HA) and reinterview the device to test your changes.

PR #6453 (2023-10-24): feat(config): add Ring Alarm Panic Button Gen2

add Ring Panic Button Gen2 to close #6440
closed 2023-10-24 by AlCalzone