zwave-js/zwave-js #6496: can not remove class

zeev-mindali, 2023-11-06:
Dear all,

i have zwave device which have a huge Scene Actuator configuration v1.
even on my rp4, it's take 30 seconds to show the value field.

[image]

so i convert the value 44 to hex which is 0x2c to remove the command class:
`
"compat": {
        "commandClasses": {
            "remove": {
                "0x2c": {
                    "endpoints": "*"
                }
            }
        }
    }
`

but it's not removing the command class.
what can be the problem?

the entire code:
{
	"manufacturer": "McoHome Co., Ltd.",
	"manufacturerId": "0x015f",
	"label": "Glass9",
	"description": "Glass 9",
	"devices": [
		{
			"productType": "0x9001",
			"productId": "0x8106"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"compat": {
	    "commandClasses": {
					"remove" : {
							"0x2e":{
								"endpoints": "*"
							},
							"0x2c":{
								"endpoints": "*"
							},
							"0x72": {
								"endpoints": "*"
							},
							"0x86": {
								"endpoints": "*"
							}
					}
	    }
	}
}

zeev-mindali, 2023-11-06:
forget to interview

closed 2023-11-06 by zeev-mindali

AlCalzone, 2023-11-06:
FYI you can also use the command class name instead of the hex ID