zwave-js/zwave-js #6497: Heatit Z-TRM3 contantly becomes unavailable and available a few seconds later

perosb, 2023-11-06:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is not described there.
- [ ] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

zwave-js-ui: 9.3.1.f26b79e

### ZwaveJS version

zwave-js: 12.2.3

### Describe the bug

Heatit Z-TRM3 contantly reports as "dead" but comes back online  just a few seconds later.

[zwavejs_current.log](https://github.com/zwave-js/zwave-js-ui/files/13259231/zwavejs_current.log)
[heatit-ztrm3.json](https://github.com/zwave-js/zwave-js-ui/files/13259232/heatit-ztrm3.json)

### To Reproduce

Unknown, I have re-included but same issue.

### Expected behavior

To be available all the time.

AlCalzone, 2024-10-08:
Sorry, this slipped through the cracks. If this is still an issue, please follow https://zwave-js.github.io/node-zwave-js/#/troubleshooting/first-steps and if that doesn't help, open a new discussion [here](https://github.com/zwave-js/node-zwave-js/discussions/new?category=request-support-investigate-issue).

closed 2024-10-08 by AlCalzone

perosb, 2024-10-08:
For completeness, I upgraded the firmware from 4.0.33 to 4.0.34 and haven't seen the issue since. OTA file https://media.heatit.com/2562?type=ota