zwave-js/zwave-js #6509: [Strips Drip 700] Some configurations have incorrect size and some incorrect name

AlCalzone, 2023-11-21:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/6508

<div type='discussions-op-text'>

<sup>Originally posted by **@dparyani** November 21, 2023</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

**What is happening?**
1. Configuration changes for [NodeId-112-0-7] and [NodeId-112-0-8] are rejected by Strips Drip 700. On closer investigation it was found that wrong configuration size is hard-coded in the JSON file.
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x019a/11_04_21_22_28.json

2. Reviewing the above mentioned file showed further incorrect discrepancies. For e.g. [NodeId-112-0-4] has incorrect configuration name.

**What did you expect to happen instead?** 
Sensative expected for these configurations to reflect and work as expected and as mentioned in the device [manual](https://staging.yggio.net/docs/strips/drip/).

**Steps to reproduce the behavior:**
  1. Pair Strips Drip 700 with Home Assistant  
  2. Enable configuration parameter [NodeId-112-0-6] Temperature Alarms.
  3. Change and queue configuration changes for parameters [NodeId-112-0-7] and [NodeId-112-0-8]. 
  4. [Manual Wake-Up](https://staging.yggio.net/docs/strips/drip/#wake-up) Strips.
  5. Try to change the environment temperature around Strips Drip 700. e.g. Use a hair dryer to increase the temperature or an ice or pre-freezed package to lower the temperature.
  6. Check the temperature levels at which the alarms trigger. They will not be similar to what the values were being set in step 3.
  7. Alternatively, one can poll for parameters [NodeId-112-0-7] and [NodeId-112-0-8]

**Anything else we should know?**
Sensative will fix this and create a pull request

### Software versions

Driver (node-zwave-js): ...

Z-Wave JS UI: 2.2.3

Home Assistant Z-Wave Core: 2023.10.5 

Home Assistant OS: 11.0

Home Assistant Z-Wave JS Addon: ...

ioBroker.zwave2 Adapter: ...

If you are using something non-standard, tell us here: ...

### Device information

**Manufacturer:** Sensative (0x019A)
**Model name:** "11 04 021/022/028"

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [ ] The log includes a re-interview of the problematic device _(if applicable)_.

- [ ] The log includes the problematic interaction with the device _(if applicable)_.

- [ ] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

_No response_</div>

dparyani, 2023-11-21:
Hello @AlCalzone,

It seems I do not have the rights to push my private branch to generate a pull request.

Attaching the affected file here...

[11_04_21_22_28.json](https://github.com/zwave-js/node-zwave-js/files/13428041/11_04_21_22_28.json)

AlCalzone, 2023-11-21:
@dparyani to create a pull request, you have to [make a fork](https://docs.github.com/en/get-started/quickstart/fork-a-repo), commit/push the changes there and then [create a pull request from your fork](https://docs.github.com/en/get-started/quickstart/contributing-to-projects#making-a-pull-request).

Doing that would be easier, since the file is automatically checked.

AlCalzone, 2024-10-30:
It appears that the file matches our current file now.